h2zero/esp-nimble-cpp #1: Sync repository with Arduino version?

mitchjs, 2020-05-10:
@h2zero I love what you are doing... 
question I don't bother with Arduino
I prefer just coding in ESP32 IDF with the C++ classes
(currently using the "other" library and gonna switch to yours)
will you be syncing all the commits from your Arduino repository to this one?
im trying to follow the changes going on right now, getting better all the time.

thanks!!!

h2zero, 2020-05-10:
Hi @mitchjs, I prefer to use IDF also, infact the original intent was to do the development here and sync it there. Just happened that the Arduino repo got a lot of interest quickly. 

Yes I will keep these in sync, forgot to push updates here last night 😞. Thanks for the reminder!

mitchjs, 2020-05-10:
great... im messing with it now... trying the way to create charestrics and descriptors
stuck on the 2904

    NimBLECharacteristic    *pMyCharacteristic = pService->createCharacteristic(CHARACTERISTIC1_UUID, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
    //myCharacteristic->setCallbacks(new myCharacteristicCallBacks());
    pMyCharacteristic->setValue("test value");

    NimBLEDescriptor        *pMyDescription    = pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2901), NIMBLE_PROPERTY::READ, 16);
  	pMyDescription->setValue("Status");

    NimBLE2904 *pMy2904 = (NimBLE2904*)pMyCharacteristic->createDescriptor("2904");
    pMy2904->setValue(NimBLE2904::FORMAT_UTF8);
`

compiler not happy with NimBLE2904 class, me doing something wrong for 2904?

h2zero, 2020-05-10:
I don't see anything wrong there, did you `#include NimBLE2904.h`?

mitchjs, 2020-05-10:
i didnt include it , but duh... its

pMy2904->**setFormat**(NimBLE2904::FORMAT_UTF8);

mitchjs, 2020-05-10:
on the include, it does need to be included... should it be? as its part of NimBLECharacteristic.cpp

h2zero, 2020-05-10:
Well I missed that one too lol, been a while since I was looking at that code, glad you found it.

To be honest I can't remember right now why it needs to be explicitly included, I think it was like that from the original library but i'm not 100% sure. You are right to question that though, all classes should be available from `NimBLEDevice.h`. When I get a chance I'll look into it.

mitchjs, 2020-05-10:
another issue...  i can connect to my new BLE device only once, just testing with lightBlue
after i disconnect,  its like not advertising anymore...

not much of a program yet... just creating a test char, and starting service and starting advertising

  	pService->start();
	NimBLEAdvertising *pAdvertising = pServer->getAdvertising();
	pAdvertising->addServiceUUID(NimBLEUUID(pService->getUUID()));
        pAdvertising->start();

do i have to do something special with Nimble?

I (3866) mycallback: onConnect()
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
I NimBLEServer: "subscribe event; cur_notify=0\n value handle; " "val_handle=8\n"
I NimBLEServer: "subscribe event; cur_notify=0\n value handle; " "val_handle=8\n"
I (20366) mycallback: onDisconnect()

h2zero, 2020-05-10:
You need to start advertising again after disconnection so the peer can find your device,
You could add `NimBLEDevice::startAdvertising();` to your `onDisconnect()` handler. Or put it in `onConnect()` and it will always advertise until you are connected to the maximum number of clients (default 3).

mitchjs, 2020-05-10:
really? didn't have to do that in the old nkolban version or it could have just been a part of Bluedroid, but ok..  im a BLEServer, i only want 1 client connecting to me at any time, which is the case now.. thanks for help, and prob need more help... hopefully my existing code just flows in... :)

h2zero, 2020-05-10:
Yeah, I the original library did that automatically in the disconnect event. I thought it should be left to the app to decide, maybe it should sleep for a while or something. I guess I should have mentioned that somewhere too 😃. Sorry about that, lots todo!

mitchjs, 2020-05-10:
:)

h2zero, 2020-05-22:
@mitchjs what do you think about the addition of  a function to enable/disable automatic advertising after disconnect? You could put it in the server creation code and wouldn’t need to implement a callback to handle that.

mitchjs, 2020-05-27:
> @mitchjs what do you think about the addition of a function to enable/disable automatic advertising after disconnect? You could put it in the server creation code and wouldn’t need to implement a callback to handle that.

yes, its a good... i think in 99% of the cases people would want the device to go back into advertising after a disconnect. i think it should default to go back into advertising

h2zero, 2020-05-28:
@mitchjs agreed, I appreciate the feedback. I'll be pushing that and some other tweaks for the server side code soon.

h2zero, 2020-07-09:
> on the include, it does need to be included... should it be? as its part of NimBLECharacteristic.cpp

> To be honest I can't remember right now why it needs to be explicitly included, I think it was like that from the original library but i'm not 100% sure. You are right to question that though, all classes should be available from `NimBLEDevice.h`. When I get a chance I'll look into it.

@mitchjs FYI the includes are no longer needed, just pushed an update to remove that requirement.

closed 2020-08-08 by mitchjs