h2zero/esp-nimble-cpp #2: Not creating Descriptors

mitchjs, 2020-05-10:
so.. 
if i create a descriptor, it doesnt see to be added to the characteristic till another one is created
example:

if i do:

```
NimBLEDescriptor        *pMyDescription    = pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2901), NIMBLE_PROPERTY::READ);
  	pMyDescription->setValue("Status");
```

by itself, it doesnt show up in LIGHTBLUE Explorer...

but if i do another one afterwards example:

```
NimBLEDescriptor        *pMyDescription1    = pMyCharacteristic->createDescriptor(DESCRIPTIOR1_UUID, NIMBLE_PROPERTY::READ);
  	pMyDescription1->setValue("test of descriptor");
```

the 1st one shows up x2901, but of course the new one doesn't

h2zero, 2020-05-10:
Are you adding this new one after you’ve started advertising or connected?

mitchjs, 2020-05-10:
nope... its all before advertising...

here is all the code, the last one doesnt show up

    // create main service
    NimBLEService *pService = pServer->createService(NimBLEUUID(SERVICE_UUID), 25);

    NimBLECharacteristic    *pMyCharacteristic = pService->createCharacteristic(CHARACTERISTIC1_UUID, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
    //myCharacteristic->setCallbacks(new myCharacteristicCallBacks());
    pMyCharacteristic->setValue("test value");

    NimBLEDescriptor        *pMyDescription    = pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2901), NIMBLE_PROPERTY::READ);
  	pMyDescription->setValue("Status");

    NimBLE2904 *pMy2904 = (NimBLE2904*)pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2904), NIMBLE_PROPERTY::READ);
    pMy2904->setFormat(NimBLE2904::FORMAT_UTF8);

    NimBLEDescriptor        *pMyDescription1    = pMyCharacteristic->createDescriptor(DESCRIPTIOR1_UUID, NIMBLE_PROPERTY::READ);
  	pMyDescription1->setValue("test of descriptor");

    // START SERVICE
  	pService->start();

	// get advertising
	NimBLEAdvertising *pAdvertising = pServer->getAdvertising();
	pAdvertising->addServiceUUID(NimBLEUUID(pService->getUUID()));

	// START ADVERISTING
	pAdvertising->start();`

h2zero, 2020-05-10:
That’s an odd issue, I don’t think any of that code has changed recently but I’ll have to test and get back to you.

h2zero, 2020-05-10:
@mitchjs Just tested and was unable to reproduce except if I used an incorrect size UUID for the descriptor. I can't see what you've defined `DESCRIPTIOR1_UUID` as, if you could post that it would be helpful.

mitchjs, 2020-05-10:
#define DESCRIPTIOR1_UUID       "11112222-0002-11e8-98d0-529269fb1459"

mitchjs, 2020-05-10:
if i change the order...

    NimBLEDescriptor        *pMyDescription    = pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2901), NIMBLE_PROPERTY::READ);
  	pMyDescription->setValue("Status");

    NimBLEDescriptor        *pMyDescription1    = pMyCharacteristic->createDescriptor(DESCRIPTIOR1_UUID, NIMBLE_PROPERTY::READ);
  	pMyDescription1->setValue("test of descriptor");

    NimBLE2904 *pMy2904 = (NimBLE2904*)pMyCharacteristic->createDescriptor(NimBLEUUID((uint16_t)0x2904), NIMBLE_PROPERTY::READ);
    pMy2904->setFormat(NimBLE2904::FORMAT_UTF8);

still the last one doesnt show up

mitchjs, 2020-05-11:
how do i change the log level to debug for the lib?

also, im using IDF v4.0.0

h2zero, 2020-05-11:
@mitchjs I was able to reproduce finally, getting to work on the fix right away. 

Log level should follow the IDF log level in Component config->Log output->Default log verbosity.

h2zero, 2020-05-11:
@mitchjs Just pushed the fix for the descriptor issue. Did you get the logs to work?

mitchjs, 2020-05-11:
i got side tracked on logs... but your fix did the trick

closed 2020-05-11 by mitchjs

mitchjs, 2020-05-11:
oh... the log output worked... :) i wanted it to be debug to help you solve the descriptor issue, but i see the change you made... dont fully understand it... but it works

h2zero, 2020-05-11:
Lol glad it's working for you, yeah that bit of code is regrettable, it's partly backward compatibility for the old library and partly a NimBLE stack thing. Thanks for the bug report though, please post an issue if you find any more 😃.