h2zero/esp-nimble-cpp #3: is the new kconfig working?

mitchjs, 2020-05-11:
is that new kconfig working? should i see it in the IDF menuconfig?

i had to remove the extension .projbuild

and then it showed up in menuconfig

[image]

h2zero, 2020-05-11:
Yes it should be working. I’ll remove that extension on the next update, was testing in idf 3.3 when I made that.

mitchjs, 2020-05-11:
just FYI
I'm testing(err programming) in Visual Studio Code with the esp-idf-extension (espressif official)
and IDF v4.0 release

h2zero, 2020-05-11:
Good to know what people are using so I can do more testing. I spend more time on that these days than coding lol. I’ve been using eclipse and idf 3.2, 3.3, 3.3.2 and idf master. However eclipse turns into a pain when using arduino component so I just started using notepad++ for quick edits.

closed 2020-08-14 by mitchjs