h2zero/esp-nimble-cpp #4: question - nimconfig.h

mitchjs, 2020-05-19:
@h2zero, question...
what would i do for a server only setup (i guess thats also called peripherial)
do i make any changes to nimconifg.h? or put a #define before I reference "NimBLEDevice.h"?
maybe add options in kconfig?

thanks for the wonderful work
-mitch

h2zero, 2020-05-19:
@mitchjs 'nimconfig.h` is not for IDF use, it's a config file I created for easy maintenance between IDF and Arduino it should not be altered unless compiling in Arduino IDE. 

That said all you need to do for your case is go into the NimBLE options in menu config and turn off observer and client roles. 

The reason for this is that turning on/off roles in menuconfig also prevents compiling that code in the NimBLE stack, so I used it in this library also to remove unnecessary code in those cases.

mitchjs, 2020-05-19:
ah.. ok... i turned off those roles... thank you

h2zero, 2020-05-19:
Hopefully saved you some flash space.

closed 2020-08-08 by mitchjs