h2zero/esp-nimble-cpp #5: consider implementing... remove all bonded devices

mitchjs, 2020-05-19:
not 100% sure where it should go...
i put it in BLEDevice class (old lib of course)

`static void		   RemoveAllBondedDevices();`

```
/* STATIC */ void BLEDevice::RemoveAllBondedDevices(void)
{
    int dev_num = esp_ble_get_bond_device_num();

	ESP_LOGI(LOG_TAG, "esp_ble_get_bond_device_num()=%d", dev_num);

    esp_ble_bond_dev_t *dev_list = (esp_ble_bond_dev_t *)malloc(sizeof(esp_ble_bond_dev_t) * dev_num);
    esp_ble_get_bond_device_list(&dev_num, dev_list);
    for (int i = 0; i < dev_num; i++) {
        esp_ble_remove_bond_device(dev_list[i].bd_addr);
    }

    free(dev_list);
}
```

h2zero, 2020-05-19:
Hi @mitchjs, I haven't yet reached the point where I'll be implementing extra support for bond maintenance etc. I hope to get there soon though. If you need this functionality right away you can use `ble_store_clear()` to erase all bonds.

mitchjs, 2020-05-19:
ah that nice and simple (thank you nimble)

finger563, 2024-05-03:
@h2zero i've noticed (on esp-idf `v5.2.1`) that the `ble_store_clear()` doesn't work as expected - I still get too many bonded device errors even though every time i enter pairing mode I call `NimBLEDevice::deleteAllBonds()`.

closed 2024-06-06 by h2zero