h2zero/esp-nimble-cpp #6: consider - the warning in "NimBLEServer.cpp"

mitchjs, 2020-05-19:
in "NimBLEServer.cpp"
just to get rid of a warning, that is OK
(my eyes always think its some error)
warning: missing initializer for member 'ble_sm_io::<anonymous>' [-Wmissing-field-initializers]
             struct ble_sm_io pkey = {0};

line 352:
`struct ble_sm_io pkey = {0};`

change to:
`struct ble_sm_io pkey = {0, 0};`

could also do:

```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wmissing-field-initializers"
struct ble_sm_io pkey = {0};
#pragma GCC diagnostic pop
```

h2zero, 2020-05-19:
Haha yes, that always annoys me too. There’s another somewhere else too that I’ve been meaning to change. Also need to get rid of the unused variable warnings when not using debug logs :). Thanks for the reminder.

h2zero, 2020-06-11:
This was fixed in #8

mitchjs, 2020-06-13:
thanks... i just pulled latest, and so many changes :)
how tested is the server
i have a 100% working bluedroid project, that i will convert soon and that will be good test for me

h2zero, 2020-06-13:
Server is fully tested and working well, very few changes were made to the API. You shouldn’t have to do much to convert your project.

Let me know if you have any problems.

closed 2020-06-22 by mitchjs