h2zero/esp-nimble-cpp #7: Empty name on first scan?

elliottkember, 2020-05-25:
I have a problem where, the first time I scan for Bluetooth LE devices from iOS, the `name` is empty.

I'm using this: `BLEDevice::init("MyDevice");` but it's coming through as "unknown" on my BLE scanning app and in iOS it comes through as `undefined`. I wonder whether I'm initializing things in the wrong order, or whether this name is not correctly saved and advertised. Any ideas?

h2zero, 2020-05-25:
The device name is pushed into the scan response packet if your advertisement data exceeds 31 bytes when the name is included. If you don't have `setScanResponse()` set to `true` (default) then the scanner won't see the name.

In my own experience also, phones usually cache the name so if it was undefined before but isn't now it might be using the cached name still, try turning bluetooth off and on and scanning again.

elliottkember, 2020-05-25:
I fixed this with:

```    NimBLEAdvertisementData advertisementData;
    advertisementData.setName("MyDevice");
    pServer->getAdvertising()->setAdvertisementData(advertisementData);
```

h2zero, 2020-05-25:
Oops sorry, I forgot that you have to set it like that as well. Glad you got it working.

elliottkember, 2020-05-25:
Same! Was a very strange bug - MAC addresses are cached by the device, so any chip that had been connected with the previous Bluedroid firmware still connected. New ESP32s that had only had NimBLE didn’t show up.

h2zero, 2020-05-26:
I'm not sure if the bluedroid library did that automatically or not, but it should be I think. I'll have a look into this, thanks!

elliottkember, 2020-05-26:
Yeah, from my testing I think it definitely did! Old firmwares using Bluedroid showed their names in a BLE scan.

h2zero, 2020-05-26:
Thanks, I’ll look into adding this.

mitchjs, 2020-05-26:
im not sure i noticed this...
i see my name of "TEST-BT!" showing up

i have 
```
// create BLE service
NimBLEDevice::init("TEST-BT!");
NimBLEServer *pServer = NimBLEDevice::createServer();
```

using nRF Connect on ios
also LightBlue Explorer

h2zero, 2020-05-26:
@mitchjs, thanks! I was just testing this here as well and not able to reproduce the issue. Looking at the code I don't see anywhere it would not set the name, except in specific cases. 

Not sure what's going on for @elliottkember here, I know he's using IDF v3.2 but the backport I made for that is equivalent to IDF master.

mitchjs, 2020-05-27:
im on IDF 4.0 for the record

h2zero, 2020-05-27:
@mitchjs I had a question for you in #1, if you get a chance I'd appreciate the input.

h2zero, 2020-06-04:
@elliottkember I think I found the problem you were having here, working on the fix along with a big server code refactoring. Should be updated soon. If you're interested in helping verify, try adding `pAdvertising->addServiceUUID(*some uuid*)` to your code before starting advertising and you should see the name.

h2zero, 2020-06-11:
This was fixed in #8

closed 2020-06-11 by elliottkember