h2zero/esp-nimble-cpp #10: would you be ok with setvalue(uint8_t& data8)?

mitchjs, 2020-06-13:
i like to have this, just so i dont have to use the one with length for a simple byte
i mean we got others (16,32...)
and it just looks nicer in user code

pCharacteristic->setValue(btBypassEnabled); 

in BLECharacteristic.cpp

```
void BLECharacteristic::setValue(uint8_t& data8) {
	uint8_t temp = uint8;
	setValue((uint8_t*)&temp, 1);
} // setValue
```

thanks
mitch

h2zero, 2020-06-13:
Yep, that's certainly a good addition.

mitchjs, 2020-06-13:
if i knew how exactly to use GIT i would do it for ya... but its so tiny... prob not worth me doing... you can paste that in and add the method to .h file
(other than git clone and git pull :) )

h2zero, 2020-06-13:
No worries, I'll take care of it 😄

mitchjs, 2020-06-19:
reminder....

h2zero, 2020-06-22:
done

closed 2020-06-22 by mitchjs