h2zero/esp-nimble-cpp #12: redefinitions - nimconfig.h

mitchjs, 2020-06-14:
i just notices after a clean and build
i noticed tons of redefines
like #pragma once not working?
not 

```
[912/942] Building CXX object esp-idf/esp-nimble-cpp/CMakeFiles/__idf_esp-nimble-cpp.dir/src/NimBLEAdvertisedDevice.cpp.obj
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:17:
../Components/esp-nimble-cpp/src/nimconfig.h:13: warning: "CONFIG_BT_NIMBLE_ENABLED" redefined
 #define CONFIG_BT_NIMBLE_ENABLED
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:14:
config/sdkconfig.h:68: note: this is the location of the previous definition
 #define CONFIG_BT_NIMBLE_ENABLED 1
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:17:
../Components/esp-nimble-cpp/src/nimconfig.h:21: warning: "CONFIG_BT_NIMBLE_ROLE_BROADCASTER" redefined
 #define CONFIG_BT_NIMBLE_ROLE_BROADCASTER
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:14:
config/sdkconfig.h:79: note: this is the location of the previous definition
 #define CONFIG_BT_NIMBLE_ROLE_BROADCASTER 1
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:17:
../Components/esp-nimble-cpp/src/nimconfig.h:29: warning: "CONFIG_BT_NIMBLE_ROLE_PERIPHERAL" redefined
 #define CONFIG_BT_NIMBLE_ROLE_PERIPHERAL
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:14:
config/sdkconfig.h:78: note: this is the location of the previous definition
 #define CONFIG_BT_NIMBLE_ROLE_PERIPHERAL 1
 
[913/942] Linking CXX static library esp-idf\button\libbutton.a
[914/942] Linking CXX static library esp-idf\wifi_provisioning\libwifi_provisioning.a
[915/942] Building CXX object esp-idf/esp-nimble-cpp/CMakeFiles/__idf_esp-nimble-cpp.dir/src/NimBLEClient.cpp.obj
In file included from ../Components/esp-nimble-cpp/src/NimBLEClient.cpp:17:
../Components/esp-nimble-cpp/src/nimconfig.h:13: warning: "CONFIG_BT_NIMBLE_ENABLED" redefined
 #define CONFIG_BT_NIMBLE_ENABLED
 
In file included from ../Components/esp-nimble-cpp/src/NimBLEClient.cpp:14:
config/sdkconfig.h:68: note: this is the location of the previous definition
 #define CONFIG_BT_NIMBLE_ENABLED 1
```

i might have screwed something up, its with IDF v4.0.1 (stable release)

is #pragma once not working???

h2zero, 2020-06-14:
Thanks for reporting, I’ll check it out. Tricky thing to balance out that code between idf versions and arduino, could definitely be a bug somewhere.

h2zero, 2020-06-14:
IDF master changed the defines to include both now, lol such a pain. Just pushed a fix for these warnings.

mitchjs, 2020-06-14:
thank you! im sure they driving you crazy

closed 2020-06-14 by mitchjs