h2zero/esp-nimble-cpp #13: security for auto created 0x2902 descriptor

mitchjs, 2020-06-15:
I want to add security for this auto created descriptor
maybe need to add NIMBLE_PROPERTY::NOTIFY_ENC?
where in code does the 0x2902 descriptor get created?

mitchjs, 2020-06-15:
or maybe a setProperty... ie

pMyCharacteristic->getDescriptorByUUID(NimBLEUUID((uint16_t)0x2902))->setProperty(NIMBLE_PROPERTY:...)

h2zero, 2020-06-15:
I’m going to look into how nimble handles this descriptor. It seems to me that it should apply the ENC_WRITE property to it if the characteristic has ENC_READ set. 

It seems possible to do this in the library, but I feel it would be better done in the stack where the descriptor is created.

mitchjs, 2020-06-15:
oh, i thought YOU(as in the cpp classes) checked if the NOTIFY property was set and you created the descriptor

h2zero, 2020-06-15:
Yes that is also done, a 2902 class instance is created but it’s basic function is to trigger read/write callbacks and keep a list of subscribed clients. Nimble takes over the rest of it including the registration of the descriptor which is where the properties are set.

We could possibly trigger security actions in the write event handler but before doing that I’d like to see if the property could be set in the stack instead.

mitchjs, 2020-06-15:
look at ble_gatts_register_clt_cfg_dsc() in ble_gatts.c (in nmible source folder)

h2zero, 2020-06-15:
Thanks! Saved me some time looking it up. I’m at work right now so I’ll have to wait for a break to look into it. Searching github on a phone is awful lol.

h2zero, 2020-06-15:
Had a good look through and it would seem the best way to go is to check the security flag of the characteristic and if the connection is encrypted before sending any notifications. The other thing that could be done is force starting the security request at that stage.

mitchjs, 2020-06-15:
yes its bad

i can connect, put in the wrong pin, turn on notifications, and then in code create a notify
and lightblue shows me the value...
big time security leak!

so.. need to get WRITE_ENC on that descriptor, so one cant turn on notifications without being bonded/authenticated

h2zero, 2020-06-15:
I saw the same thing in my testing. The code you mentioned above does not accommodate the security setting. I think I have a solution, I I started working on it in the the new notify-security branch I just pushed. Should have a full solution tonight.

mitchjs, 2020-06-15:
woo... i need to get back to converting my actual project over... then it will be real good test of lib

h2zero, 2020-06-22:
I think this one is resolved for now.

mitchjs, 2020-06-22:
so.. in that ESP-IDF NimBLE periperial demo... i enabled bonding/encryption 
and added a NOTIFY property to one of the demo characterisctics, and sure enough
you can enable/disable the subscription without being secure/bonded

i didnt test if the value would get sent back... not sure how to do that in the nimble native api
but i suspect it would have...
maybe time to open issue at mynewt/nimble

h2zero, 2020-06-22:
Yes, it needs to be brought up there. I wanted to look into the code more to see how it should be handled but if you want to open the issue there I will chime in on it later.

Also, just so you know, you can manually call `NimBLEDevice::startSecurity()` in the onConnect() callback if you just want to make sure the whole device is secured. Then disconnect clients on failure.

mitchjs, 2020-06-22:
ah maybe thats my solution, force the app to bond at connect, not have to worry if they access notfication 1st, which im not sure what they do 1st... i didnt write it :)

mitchjs, 2020-06-22:
to do the NimBLEDevice::startSecurity(), it needs the connection id
i have of course from the callback NimBLEServer* pServer

what do i need to do to get connection id?

thanks

h2zero, 2020-06-22:
You can do it like this:
```
 void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
        NimBLEDevice::startSecurity(desc->conn_handle);
```

mitchjs, 2020-06-22:
ah i didnt know we had an overloaded callback :)

h2zero, 2020-06-23:
Oh, oops. that should be public. I'll change it.

closed 2020-08-08 by mitchjs