h2zero/esp-nimble-cpp #19: fatal error: nimble/ble.h: No such file or directory

Dave-Osborne, 2020-07-17:
Hi,

I'm trying to swap to nimble but am getting the following error..
lib/esp-nimble/src/NimBLEAddress.h:20:24: fatal error: nimble/ble.h: No such file or directory

I'm using PlatformIO (Arduino as a component of ESPIDF) and have copied this repo. and the esp-nimble-component repo. to the lib folder.

Versions used are  
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - framework-espidf 3.30300.190916 (3.3.0) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)

I wasn't sure if I needed the nimble sections or not in SDKCONFIG.H so have just generated and included those, but still not luck.

I really want to use nimble with my project and would appreciate any help.

Thanks!

h2zero, 2020-07-17:
Hi @Dave-Osborne could you describe your directory structure for your project? 

> copied this repo. and the esp-nimble-component repo. to the lib folder.

I believe this could be the cause, as this repo needs to be in the project/components folder, not in the arduino libraries folder.

Directory structure should look like :
```
project folder
       - main
              - main.cpp
              - libraries
       - components
              - arduino
              - esp-nimble-cpp
              - esp-nimble-component
```

Dave-Osborne, 2020-07-20:
Thanks for the heads up @h2zero - I'm using PlatformIO which has a different folder structure and it looks like the version I'm using doesn't allow espidf components to be built.

I'll take a look at your Arduino version instead.

closed 2020-07-20 by Dave-Osborne

h2zero, 2020-07-20:
Sorry I guess I missed that. Yes the arduino version works with platformio.