h2zero/esp-nimble-cpp #24: 0x2902 Help - subscribing to notifications

mitchjs, 2020-07-25:
@h2zero, really need your help
i created a service that has 10 characteristics
of which 7 of them support notify

while testing in light blue, if i start turning on subscribe for each one, i get to like the 5th
and lightblue reports "failed to write descriptor properly (status 6)"

nimble says:

I NimBLEServer: "subscribe event; attr_handle=33, subscribed: True"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
I NimBLECharacteristic: "Characteristic UUID: ec827d85-a1c1-11e8-98d0-529269fb1459 subscribed" (i added)

the notfications dont work on the ones that give this error (always 5th one)
where can i check that the descriptor 0x2902 value is set
could we be returning a error back to the client?

ps im still running code with the 2902 class

mitch

mitchjs, 2020-07-25:
yikes, this could be an ANDROID problem.... i just tried lightblue on my iphone
and i was able to subscribe to all 7 of my notification
my NEW tablet is running android 9!

h2zero, 2020-07-25:
Are you bonded with the device when this happens? There is a bug in that code that was just patched 3 days ago in esp-nimble, having to do with cccd storage.

mitchjs, 2020-07-25:
ok, i think there is a problem releated...

all in lightblue ios
if i disconnect the device, then reconnect... and do nothing else
nimble:

GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (191912) DS-EZLOCK-ECM9(BLE): BT - connected
I (192162) DS-EZLOCK-ECM9(BLE): BT - onAuthenticationComplete() - ENCRYPTED
I (192162) DS-EZLOCK-ECM9(BLE): BT - onAuthenticationComplete() - BONDED
I (192172) DS-EZLOCK-ECM9(BLE): BT - onAuthenticationComplete() - AUTHENTICATED
I NimBLEServer: "subscribe event; attr_handle=19, subscribed: True"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
I NimBLECharacteristic: "Characteristic UUID: ec827d81-a1c1-11e8-98d0-529269fb1459 subscribed"
I NimBLEServer: "subscribe event; attr_handle=28, subscribed: True"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
I NimBLECharacteristic: "Characteristic UUID: ec827d84-a1c1-11e8-98d0-529269fb1459 subscribed"
I NimBLEServer: "subscribe event; attr_handle=33, subscribed: True"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
I NimBLECharacteristic: "Characteristic UUID: ec827d85-a1c1-11e8-98d0-529269fb1459 subscribed"
I NimBLEServer: "subscribe event; attr_handle=38, subscribed: True"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
I NimBLECharacteristic: "Characteristic UUID: ec827d83-a1c1-11e8-98d0-529269fb1459 subscribed"
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: False"

WHY is it subscribing? these where what i did subscribe before the connection
is it remembering what i WAS subscribed too?
is that right?? i think not... 

infact lightblue doesnt thing its subscribed, i have to press listen, which does nothing, and then stop the listen just to clear it

mitchjs, 2020-07-25:
> Are you bonded with the device when this happens? There is a bug in that code that was just patched 3 days ago in esp-nimble, having to do with cccd storage.

yes... im bonded..

mitchjs, 2020-07-25:
yikes... i did git pull on esp-nimble... and whole new problems!
i connect, i pair, it bonds...
i disconnect....
i reconnect, cant get in
all broken

mitchjs, 2020-07-25:
on the plus side, my initial issue is GONE
i was able to enable the notifications from all 7 chars now

h2zero, 2020-07-25:
The reason you were getting those subscribe events is because you are bonded. Nimble stores the status of all cccds when a bonded device subscribes. When you reconnect it tells the app about them. Light blue doesn’t store this same info, but it should. If you have a custom app you should store those also.

h2zero, 2020-07-25:
I should note that the storage of cccds is part of the Bluetooth spec when it comes to bonded devices. LightBlue is just not following it.

mitchjs, 2020-07-25:
ok.. i think the app is happy now... 

ah didnt know that about the cccd storage... if the spec says save em, then lets save em
my guess is the app tries to subscribe, and its subscribed already no harm

im doing tests, but the nimble update seemed to solve my original issue

h2zero, 2020-07-25:
Glad to hear it!

mitchjs, 2020-07-25:
closing issue

closed 2020-07-25 by mitchjs