h2zero/esp-nimble-cpp #34: NimBLEDevice::init() call results in boot loop

ntwallace, 2021-01-26:
I finally took the plunge and upgraded my codebase from IDF 3.3 to 4.2, and everything works except my BLE code. I now get a `rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)` boot loop on device start unless I comment out NimBLEDevice::init() from my code. The boot loop happens before the app CPU starts, so no code is actually being run. 

The NimBLE settings in SDKConfig are exactly the same as I had on 3.3, any thoughts on what might cause this? Changing the memory allocation strategy from default to internal or SPI RAM makes no difference. Full debug boot log:

```cpp
D (1001) efuse: coding scheme 0
/Users/nick/esp-idf/components/freertos/queue.c:1441 (xQueueSemaphoreTake)- assert failed!

abort() was called at PC 0x40093854 on core 0
0x40093854: xQueueSemaphoreTake at /Users/nick/esp-idf/components/freertos/queue.c:1445 (discriminator 1)

Backtrace:0x4008fe1a:0x3ffe3990 0x400905d1:0x3ffe39b0 0x400962f2:0x3ffe39d0 0x40093854:0x3ffe3a40 0x40094cc6:0x3ffe3a80 0x4022d4a5:0x3ffe3aa0 0x40094c65:0x3ffe3ad0 0x4022506a:0x3ffe3b20 0x40224f3d:0x3ffe3b70 0x40225205:0x3ffe3b90 0x4008dfec:0x3ffe3bc0 0x401ea7ba:0x3ffe3c30 0x400810ee:0x3ffe3c50 0x40079202:0x3ffe3c80 |<-CORRUPTED
0x4008fe1a: panic_abort at /Users/nick/esp-idf/components/esp_system/panic.c:367

0x400905d1: esp_system_abort at /Users/nick/esp-idf/components/esp_system/system_api.c:108

0x400962f2: abort at /Users/nick/esp-idf/components/newlib/abort.c:46

0x40093854: xQueueSemaphoreTake at /Users/nick/esp-idf/components/freertos/queue.c:1445 (discriminator 1)

0x40094cc6: esp_log_impl_lock_timeout at /Users/nick/esp-idf/components/log/log_freertos.c:45

0x4022d4a5: esp_log_writev at /Users/nick/esp-idf/components/log/log.c:170

0x40094c65: esp_log_write at /Users/nick/esp-idf/components/log/log.c:199

0x4022506a: esp_efuse_utility_process at /Users/nick/esp-idf/components/efuse/src/esp_efuse_utility.c:71 (discriminator 9)

0x40224f3d: esp_efuse_read_field_blob at /Users/nick/esp-idf/components/efuse/src/esp_efuse_api.c:47

0x40225205: esp_efuse_get_pkg_ver at /Users/nick/esp-idf/components/efuse/src/esp32/esp_efuse_fields.c:65

0x4008dfec: psram_enable at /Users/nick/esp-idf/components/esp32/spiram_psram.c:808

0x401ea7ba: esp_spiram_init at /Users/nick/esp-idf/components/esp32/spiram.c:150

0x400810ee: call_start_cpu0 at /Users/nick/esp-idf/components/esp_system/port/cpu_start.c:337
```

ntwallace, 2021-01-26:
🤦‍♂️🤦‍♂️🤦‍♂️ this is a repeat of what I described [here](https://github.com/h2zero/NimBLE-Arduino/issues/32#issuecomment-701038305) last September. It appears that 4.2 by default has a greater DRAM footprint, and was causing this same issue. Closing.

closed 2021-01-26 by ntwallace

h2zero, 2021-01-26:
At least you found the problem quickly, yes 4.2 uses more resources and may require some increases to task stack size in your app.