h2zero/esp-nimble-cpp #35: No support for multiple characteristics with same UUID in one service

DavidJRobertson, 2021-02-07:
The bluetooth spec allows having duplicate characteristics within a service provided there is a difference in descriptor values. For example, see the spec for environmental sensing service (https://www.bluetooth.com/specifications/specs/environmental-sensing-service-1-0/):

"The Server may expose multiple instances of the same ESS Characteristic (i.e., characteristics with the same UUID) provided that each instance has an ES Measurement descriptor (see Section 3.1.2.1) and each ES Measurement descriptor has a unique value to allow the instances to be differentiated by the Client. This allows a sensor to report its readings in more than one way. For example, a temperature sensor may provide both instantaneous and daily average readings. In this example, the instantaneous and average measurements are identified by the ES Measurement descriptor associated with each instance of the ESS Characteristic so the Client can discriminate between the characteristic for average temperature and instantaneous temperature."

This library does not support this use case.

h2zero, 2021-02-07:
Hi @DavidJRobertson, thank you for bringing this up. Yes, there is no simple way to accomplish this presently, though it is possible via characteristic handles in a workaround fashion. 

This is something I have planned to implement but have not yet thought of the best way to do so. I am open to suggestions if you have any to share.

DavidJRobertson, 2021-02-07:
It might be sufficient to provide a method on `NimBLEService` to get all characteristics that are defined. Or even better, a method that returns a list of characteristics matching a given UUID. And then the warning log on creating a duplicate characteristic should probably go.

I haven't seen any code that would break on adding duplicate characteristics but I might be missing something.

Another thing, which is not strictly related, is that I don't see a way to change the list characteristics of a service dynamically at runtime. I tried just creating new characteristics and then calling `start()` on the service again, but this did not appear to actually work (`E NimBLEService: "ble_gatts_add_svcs, rc= 15, "`).

h2zero, 2021-02-07:
Thanks for your feedback, it would be simple enough to implement a `getCharacteristics` method in `NimBLEService` that could return the vector of characteristics belonging to the service. Also I think it would be manageable to add an index parameter to `getCharacteristic` such that a reference could be provided for the UUID and an instance number in the multiple same UUID scenario. Please let me know your thoughts.

> Another thing, which is not strictly related, is that I don't see a way to change the list characteristics of a service dynamically at runtime. I tried just creating new characteristics and then [...]

This is a little trickier, for that use case to work the gatt server needs to be completely reset and restarted. It requires the NimBLE host to change the handles of every attribute to accomplish this. It may not be too difficult to implement however, I will look to add this when I do the server code pass that I'm planning soon.

closed 2021-02-10 by DavidJRobertson