h2zero/esp-nimble-cpp #37: Build errors with ESP-IDF 4.3

BarakChamo, 2021-02-09:
Hi @h2zero 👋

Thanks for your great work on this component!

I recently upgraded a project to IDF 4.3, and trying to build with this component at 1.2.0.
My project partners are able to build just fine with the same setup, running IDF's master branch, but I'm getting a bunch of errors from the component and wanted to ask if you'd have any pointers on how to fix them.

This is the problem view from VS Code:
[image]

And details from some of the errors, that seem to repeat:
```bash
In file included from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:24,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/hci_common.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:30,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:31,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.h:23,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:23:
c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\string.h:31:17: note:   initializing argument 1 of 'void* memcpy(void*, const void*, size_t)'
 void *  memcpy (void *__restrict, const void *__restrict, size_t);
                 ^~~~~~
../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:525:69: error: assignment of member 'ble_uuid_t::type' in read-only object
                 m_advData.uuids128[m_advData.num_uuids128].u.type = BLE_UUID_TYPE_128;
                                                                     ^~~~~~~~~~~~~~~~~
../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:602:28: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
             free(m_advData.uuids128);
                  ~~~~~~~~~~^~~~~~~~
In file included from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\cstdlib:75,
                 from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\stdlib.h:36,
                 from C:/Users/Barak/esp/esp-idf/components/newlib/platform_include/assert.h:21,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:26,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/hci_common.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:30,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:31,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.h:23,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:23:
c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:97:12: note:   initializing argument 1 of 'void free(void*)'
 void free (void *) _NOTHROW;
            ^~~~~~
../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:608:28: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
             free(m_advData.uuids32);
                  ~~~~~~~~~~^~~~~~~
In file included from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\cstdlib:75,
                 from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\stdlib.h:36,
                 from C:/Users/Barak/esp/esp-idf/components/newlib/platform_include/assert.h:21,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:26,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/hci_common.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:30,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:31,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.h:23,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:23:
c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:97:12: note:   initializing argument 1 of 'void free(void*)'
 void free (void *) _NOTHROW;
            ^~~~~~
../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:614:28: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
             free(m_advData.uuids16);
                  ~~~~~~~~~~^~~~~~~
In file included from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\cstdlib:75,
                 from c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\stdlib.h:36,
                 from C:/Users/Barak/esp/esp-idf/components/newlib/platform_include/assert.h:21,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:26,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/hci_common.h:23,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:30,
                 from C:/Users/Barak/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:31,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.h:23,
                 from ../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:23:
c:\users\barak\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:97:12: note:   initializing argument 1 of 'void free(void*)'
 void free (void *) _NOTHROW;
```

I followed your instructions on settings up Nimble in the config, which worked in previous builds but now is failing to compile.

I'd appreciate any pointers!

h2zero, 2021-02-09:
Hi @BarakChamo, you're welcome 😄 

Sorry to hear of this issue, I'm not sure why it's occurring for you, I have not seen this in my testing but I have not compiled against 4.3. I'm not sure what the cause is but I will look into it.

BarakChamo, 2021-02-09:
@h2zero thanks!

h2zero, 2021-02-09:
Just did a quick test and can confirm the issue. It doesn't seem to be caused by any changes in this repo, seems something has changed with the nimble core files. I'll try to get this fixed soon.

BarakChamo, 2021-02-09:
ok glad you're able to reproduce this at least, thanks for your help!

h2zero, 2021-02-09:
Well that was an easy find, they changed IDF to use nimble 1.3.0 instead of 1.2.0 prior. Now many parameters have const qualifiers. Looks like I have some work to do.

h2zero, 2021-02-10:
@BarakChamo just pushed a commit that fixes this.

BarakChamo, 2021-02-10:
@h2zero you're the best! 
Will check right now and confirm.

BarakChamo, 2021-02-10:
Works! Thank you sir!

closed 2021-02-10 by BarakChamo