h2zero/esp-nimble-cpp #40: Gatt connection timeout

rweller59, 2021-03-20:
I am migrating my esp32 code to NimBle and discovered your library. I have used cpp_utils in the past and appreciate you extending them for NimBle.

I am getting a Gatt Conn Timeout error with my code. Went back to your example code and found the same using both my code and nRF Connect.

I basically just scan and connect. Any ideas on what I might look for?

h2zero, 2021-03-20:
Hi @rweller59, could you describe further the circumstances of this error? It's not clear as to what the setup is, are you connecting as a client or advertising / server, or both? 

Would you mind enabling debug logging and post the results? That would be far quicker for me to source the issue.

rweller59, 2021-03-20:
sorry for the lack of detail.. this is from test code using your example examples/advanced/nimble_server. log file is output connecting 2 times from nRF Connect using Samsung Note 5.

The timing of the disconnect is not consistent, anywhere from a couple of seconds to about 20 secs. Log looks the same when using S7. Wondering if it has anything to do with age of phone but unfortunately I do not have access to anything newer at the moment.
[debug_log.txt](https://github.com/h2zero/esp-nimble-cpp/files/6175557/debug_log.txt)

h2zero, 2021-03-20:
Thanks for the log, something interesting is the reason code:

0x08 | BLE_HS_ENOTSUP | Operation disabled at compile time.
-- | -- | --

Also the timeout is quite low in the example, I would suggest removing the call to `updateConnParams` in the connect callback and let the phone dictate the parameters to see what happens.

rickAllDev, 2021-03-20:
removed updateConnParams call, 
modified parms (updateConnParams(desc->conn_handle, 0, 0x20, 0x10, 400);) based on another discussion thread (https://github.com/espressif/esp-idf/issues/3551)

neither change seemed to make a difference.

h2zero, 2021-03-20:
Sorry, hadn't had my coffee yet, forgot to translate the reason code to HCI which would be:

0x0208 | 0x08 | BLE_ERR_CONN_SPVN_TMO | Connection Timeout
-- | -- | -- | --

As you tested without the call we know that isn't the issue then. However, just so you are aware, the parmeter order in that call is different from bluedroid so `updateConnParams(desc->conn_handle, 0, 0x20, 0x10, 400);` would be invalid parameters in NimBLE; [details here](https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_server.html#a9b08738766de90d7065263761314030a).

Issues involving timeouts are very difficult to resolve without a sniffer trace, that leaves us with trial and error. I suggest leaving the `updateConnParams` call deleted so that can stay ruled out. 

The next thing we can try is to advertise the connection parameters so the phone will use them on connection, this can be done by adding before `pAdvertising->start();` :
```
pAdvertising->setMinPreferred(36);
pAdvertising->setMaxPreferred(48);
```

If you could try that and post another log if it fails that would help greatly. Also If you could share your sdkconfig I will try to reproduce here.

rickAllDev, 2021-03-20:
sorry, totally missed the parameter order.
added pAdvertising.  same result. attached is log and sdkconfig
[debug_log.txt](https://github.com/h2zero/esp-nimble-cpp/files/6175870/debug_log.txt)
[sdkconfig.txt](https://github.com/h2zero/esp-nimble-cpp/files/6175871/sdkconfig.txt)

I'm obviously not very knowledgeable re: ble. Why would timeouts vary from your setup to mine?

h2zero, 2021-03-20:
Something is strange in your log as I can see `updateConnParams` is still being called. Please check that [this lineis removed](https://github.com/h2zero/esp-nimble-cpp/blob/7dd4d6880632d7a75c52a952b710bb1ce58c626f/examples/Advanced/NimBLE_Server/main/main.cpp#L38)

Also I can see it actually was working fine in one of the connection events when the parameters were different (10 second timeout).

>Why would timeouts vary from your setup to mine? 

They shouldn't, I just like to see if there is some config variable that may be causing issues.

rickAllDev, 2021-03-20:
double-checked, definitely commented out. attached code along with another run log. only using nrf connect at this point.
[main (copy).txt](https://github.com/h2zero/esp-nimble-cpp/files/6175955/main.copy.txt)
[debug_log.txt](https://github.com/h2zero/esp-nimble-cpp/files/6175956/debug_log.txt)

h2zero, 2021-03-20:
Not sure if you copied the correct code you are using but that file does not have the call commented. 
I have modified your file, provided below. Sorry if there is some confusion.
[main_modified.txt](https://github.com/h2zero/esp-nimble-cpp/files/6175978/main_modified.txt)

rickAllDev, 2021-03-20:
I had main.cpp from the examples opened and that is what I had been editing since you suggested I remove updateConnParams. Just commented the line in the actual main.cpp and have been connected for 10 minutes without disconnect. So, looks like that resolved the issue? Sorry about wasting your time.

Went back to my app, made the change, and no more disconnects there (at least for past 5 min).

In general, does it make more sense to let the connecting client define these connect parms rather than setting as was done in the example?

h2zero, 2021-03-20:
Great, glad to hear it's working 😄 

>In general, does it make more sense to let the connecting client define these connect parms rather than setting as was done in the example?

Yes I highly recommend that approach unless there is a reason to change them, i.e big data transfer so faster interval etc. This exists in the example merely for demonstration purposes and not intended for regular use. I should make a note of that in the example.

rickAllDev, 2021-03-20:
Ok. Thanks for the quick support on this and add'l input.

closed 2021-03-20 by rickAllDev