h2zero/esp-nimble-cpp #43: minor issue: can't turn off advertising log messages

rickAllDev, 2021-03-28:
I'm toggling advertising on and off and get log messages every toggle:
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
GAP procedure initiated: stop advertising.

code:
class task_advertising: public Task {
  void run(void *data) {
    NimBLEAdvertising *advertising_t = NimBLEDevice::getAdvertising();
		while(true) {
      while(!ble_connected) {
			  if (!advertising_t->isAdvertising()) 
          advertising_t->start();
          //NimBLEDevice::stopAdvertising();
          else
          //NimBLEDevice::startAdvertising();
          advertising_t->stop();
			  delay(1000);
	    }
    delay(1000);
    }
}};

DavidJRobertson, 2021-03-28:
Related to my issue #41

h2zero, 2021-03-28:
Hi @rickAllDev, I just merged the PR by @DavidJRobertson that should help with this, could you give that a try and let me know?

h2zero, 2021-03-29:
It appears there is a new log config for NimBLE in the latest idf, assuming that's what you're using. If so you can change the NimBLE log level to none in menuconfig and those messages will be gone.

mickeyl, 2023-04-28:
Can we close this?

closed 2024-06-06 by h2zero