h2zero/esp-nimble-cpp #44: Dynamically adding characteristics to services

thomcrielaardBCU, 2021-05-20:
Hey everyone!

For a project I need to dynamically add characteristics to a service after the service has started. I have no way of telling how much characteristics are remaining, so counting the characteristics before starting a service won't be a possibility. 

I feel like the only way to achieve this is to create a new service, copy the existing characteristics to the new service, delete the old service and start the new service. However, the problem is that the getProperties function for the characteristics is private. This makes it impossible for me to copy the characteristics, since I have no way of telling which properties the characteristics have. 

Is there a different way to achieve this? And if not, should I make a pull request to fix this issue?

Thanks in advance,

Thom

thomcrielaardBCU, 2021-05-20:
If you are currently working on it I will just fix it locally for now and update the library once it is finished. Thank you for your reply!

closed 2021-05-20 by thomcrielaardBCU

h2zero, 2021-07-20:
@thomcrielaardBCU Sorry for the delay. I have created PR #54 to address this request. If you have a chance to try it please provide feedback :).

thomcrielaardBCU, 2021-07-20:
@h2zero I will test it after my holidays! Thank you very much! :)

h2zero, 2021-07-21:
Enjoy your holidays! I will merge this shortly as it works well in my testing. If anything needs to be changed we can do it in a future patch.