h2zero/esp-nimble-cpp #45: Lower Memory footprint

0xFEEDC0DE64, 2021-05-28:
Hi,
I added the simple example to our existing firmware code and sadly, nimble seems to eat up all the free memory?

when I start adding more and more characteristics, will this go even more down?

(Before the first outage, bluetooth was completely disabled in sdkconfig and nimble was not linked to the application, after the short break all the memory is gone)
[image]

h2zero, 2021-05-28:
Hi @0xFEEDC0DE64, I am assuming you are also using wifi in this project? NimBLE is very small on it's own but when the co-exist libraries are linked as well there is significantly more RAM consumed (still much less than with bluedroid). This is sadly something that is out of our control as it is a part of IDF. Heap use can be reduced with NimBLE config settings in sdkconfig, turning off the BLE roles you are not using and reducing the max connections and and memory block sizes will help.

0xFEEDC0DE64, 2021-05-30:
WiFi was used at the same time, and i disabled all unneeded ble features with similar ram results, I can register 1 or 2 characteristics before everything crashes down. Looks like we won't add bluetooth or ble soon in our products (at least not without a second esp32 on the board :) )

closed 2021-05-30 by 0xFEEDC0DE64

h2zero, 2021-05-30:
That's unfortunate. I don't know what all is in your device for memory consumption. but the benchmark I have for basic BLE and WiFi combination there is 170k heap leftover. If that's no enough for the project then I'd suggest a PSRAM module instead of another esp32 :)