h2zero/esp-nimble-cpp #46: BLE connection times out immediately when using ULP to wake up

AlexHartford, 2021-06-01:
Hi there, we are very new to ESP32 and IOT in general. We have successfully used this library to connect to various devices via bluetooth, and are able to send/receive data. However, after [adding some assembly code](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-ulp-pulse/ulp) to achieve a ULP mode, attempting to connect to the device using bluetooth yields an immediate disconnect or timeout.

We are hoping you may have any insight as to why this addition could be breaking our BLE functionality. [This is a minimal example with most of our functionality stripped out](https://github.com/AlexHartford/esp32-ble-test).

h2zero, 2021-06-01:
Hi @AlexHartford, it is my understanding that using the radio is not possible with the ULP.

AlexHartford, 2021-06-01:
@h2zero We're not actually trying to use the bluetooth _while in ULP_, but rather after waking up. In the example, ULP is there for a power button and while the device is powered on (not in ULP mode) is when we are facing the bluetooth issues. 

Thanks for the quick reply!

h2zero, 2021-06-02:
Ahh ok, I didn't see any deinit of the ble stack in the code so I assumed that is what you were doing, sorry.

What you'll need to do is call `NimBLEdevice::deinit(true);` before entering low power mode and then reinitialize after wake up.

That should solve the issue.

AlexHartford, 2021-06-02:
We added in deinit prior to sleep. It didn't fix the issue unfortunately. To clarify, it won't connect at any time, not only after entering deep sleep. We attached the serial monitor and phone connection logs in case that helps figure this out. (serial monitor is first, phone is second).

Serial monitor:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7284
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 20:31:30␛[0m
␛[0;32mI (29) boot: chip revision: 3␛[0m
␛[0;32mI (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0␛[0m
␛[0;32mI (40) boot.esp32: SPI Speed      : 80MHz␛[0m
␛[0;32mI (45) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (49) boot.esp32: SPI Flash Size : 16MB␛[0m
␛[0;32mI (54) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (59) boot: Partition Table:␛[0m
␛[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00005000␛[0m
␛[0;32mI (77) boot:  1 otadata          OTA data         01 00 0000e000 00002000␛[0m
␛[0;32mI (85) boot:  2 app0             OTA app          00 10 00010000 00730000␛[0m
␛[0;32mI (92) boot:  3 app1             OTA app          00 11 00740000 00730000␛[0m
␛[0;32mI (100) boot:  4 spiffs           Unknown data     01 82 00e70000 00006000␛[0m
␛[0;32mI (107) boot:  5 fctry            WiFi data        01 02 00e76000 00004000␛[0m
␛[0;32mI (115) boot: End of partition table␛[0m
␛[0;32mI (119) boot: No factory image, trying OTA 0␛[0m
␛[0;32mI (124) boot_comm: chip revision: 3, min. application chip revision: 0␛[0m
␛[0;32mI (131) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11d54 ( 73044) map␛[0m
␛[0;32mI (164) esp_image: segment 1: paddr=0x00021d7c vaddr=0x3ffbdb60 size=0x03440 ( 13376) load␛[0m
␛[0;32mI (169) esp_image: segment 2: paddr=0x000251c4 vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (171) esp_image: segment 3: paddr=0x000255d0 vaddr=0x40080404 size=0x0aa48 ( 43592) load␛[0m
␛[0;32mI (196) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x5ea00 (387584) map␛[0m
␛[0;32mI (321) esp_image: segment 5: paddr=0x0008ea28 vaddr=0x4008ae4c size=0x0b264 ( 45668) load␛[0m
␛[0;32mI (338) esp_image: segment 6: paddr=0x00099c94 vaddr=0x400c0000 size=0x00064 (   100) load␛[0m
␛[0;32mI (351) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (364) boot: Set actual ota_seq=1 in otadata[0]␛[0m
␛[0;32mI (365) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (365) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (368) cpu_start: Application information:␛[0m
␛[0;32mI (373) cpu_start: Project name:     esp32-ble-test␛[0m
␛[0;32mI (378) cpu_start: App version:      2a53651␛[0m
␛[0;32mI (383) cpu_start: Compile time:     Jun  1 2021 20:30:46␛[0m
␛[0;32mI (389) cpu_start: ELF file SHA256:  904d254efe1d21d5...␛[0m
␛[0;32mI (395) cpu_start: ESP-IDF:          3.40200.210118␛[0m
␛[0;32mI (401) cpu_start: Starting app cpu, entry point is 0x40081324␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (411) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (418) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (424) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM␛[0m
␛[0;32mI (430) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM␛[0m
␛[0;32mI (436) heap_init: At 3FFC3C00 len 0001C400 (113 KiB): DRAM␛[0m
␛[0;32mI (443) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (449) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (455) heap_init: At 400960B0 len 00009F50 (39 KiB): IRAM␛[0m
␛[0;32mI (462) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (479) spi_flash: detected chip: gd␛[0m
␛[0;32mI (480) spi_flash: flash io: dio␛[0m
␛[0;32mI (480) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
Starting main
Not ULP wakeup, initializing main prog
Starting NimBLE Server
␛[0;32mI (3544) BTDM_INIT: BT controller compile version [a4c513b]␛[0m
␛[0;32mI (3544) system_api: Base MAC address is not set␛[0m
␛[0;32mI (3544) system_api: read default base MAC address from EFUSE␛[0m
[... 7 lines cut ...]
␛[0;32mI (3954) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (3964) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
Button thread created...
Before main loop...
Client connected
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
Client address: 61:16:5e:47:7c:4a
GAP procedure initiated: connection parameter update; conn_handle=0 itvl_min=24 itvl_max=48 latency=0 supervision_timeout=120 min_ce_len=0 max_ce_len=0
Client disconnected - start advertising
␛[0;33mW (28434) NimBLEAdvertising: Advertising already active␛[0m
␛[0;33mW (28434) NimBLEAdvertising: Advertising already active␛[0m
```

Phone:
```
Connecting to 10:52:1C:88:6C::AE...
[Server] Device with address 10:52:1C:88:6C::AE connected
Connected to 10:52:1C:88:6C::AE
Connection parameters updated (interval: 60.0ms, latency: 0, timeout: 1200ms)
Discovering Services...
Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
Error 8 (0x8) GATT CONN TIMEOUT
Disconnected
[Server] Device disconnected
```

h2zero, 2021-06-02:
Looks like an issue with conflicting connection parameters.

Try removing `pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 120);` from the onconnect handler.

AlexHartford, 2021-06-02:
After removing that line, here are the logs from the phone:
```
Connecting to 10:52:1C:88:6C::AE...
[Server] Device with address 10:52:1C:88:6C::AE connected
Connected to 10:52:1C:88:6C::AE
Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
Discovering Services...
[Server] Device disconnected
Error 8 (0x8) GATT CONN TIMEOUT
Disconnected
```

h2zero, 2021-06-02:
I can't think of anything else that would cause this. 

Could you turn up the log verbosity in menuconfig to debug and share the output when it disconnects?

AlexHartford, 2021-06-02:
<details>
<summary>Device logs - Click to expand</summary>

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:9388
load:0x40078000,len:14056
load:0x40080400,len:4956
entry 0x40080730
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 16:59:05␛[0m
D (29) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)␛[0m
␛[0;32mI (36) boot: chip revision: 3␛[0m
␛[0;32mI (40) boot_comm: chip revision: 3, min. bootloader chip revision: 0␛[0m
D (47) boot.esp32: magic e9␛[0m
D (49) boot.esp32: segments 04␛[0m
D (53) boot.esp32: spi_mode 02␛[0m
D (56) boot.esp32: spi_speed 0f␛[0m
D (59) boot.esp32: spi_size 04␛[0m
␛[0;32mI (62) boot.esp32: SPI Speed      : 80MHz␛[0m
␛[0;32mI (67) boot.esp32: SPI Mode       : DIO␛[0m  
␛[0;32mI (71) boot.esp32: SPI Flash Size : 16MB␛[0m
D (76) boot: Enabling RTCWDT(9000 ms)␛[0m
␛[0;32mI (80) boot: Enabling RNG early entropy source...␛[0m
D (85) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0␛[0m
D (94) boot: mapped partition table 0x8000 at 0x3f408000␛[0m
D (99) flash_parts: partition table verified, 7 entries␛[0m
␛[0;32mI (105) boot: Partition Table:␛[0m
␛[0;32mI (108) boot: ## Label            Usage          Type ST Offset   Length␛[0m
D (116) boot: load partition table entry 0x3f408000␛[0m
D (121) boot: type=1 subtype=2␛[0m
␛[0;32mI (124) boot:  0 nvs              WiFi data        01 02 00009000 00005000␛[0m
D (131) boot: load partition table entry 0x3f408020␛[0m
D (136) boot: type=1 subtype=0␛[0m
␛[0;32mI (139) boot:  1 otadata          OTA data         01 00 0000e000 00002000␛[0m
D (147) boot: load partition table entry 0x3f408040␛[0m
D (152) boot: type=0 subtype=10␛[0m
␛[0;32mI (155) boot:  2 app0             OTA app          00 10 00010000 00730000␛[0m
D (163) boot: load partition table entry 0x3f408060␛[0m
D (168) boot: type=0 subtype=11␛[0m
␛[0;32mI (171) boot:  3 app1             OTA app          00 11 00740000 00730000␛[0m
D (178) boot: load partition table entry 0x3f408080␛[0m
D (183) boot: type=1 subtype=82␛[0m
␛[0;32mI (187) boot:  4 spiffs           Unknown data     01 82 00e70000 00006000␛[0m
D (194) boot: load partition table entry 0x3f4080a0␛[0m
D (199) boot: type=1 subtype=2␛[0m
␛[0;32mI (202) boot:  5 fctry            WiFi data        01 02 00e76000 00004000␛[0m
␛[0;32mI (210) boot: End of partition table␛[0m
D (214) boot: OTA data offset 0xe000␛[0m
D (218) bootloader_flash: mmu set paddr=00000000 count=1 size=2000 src_addr=e000 src_addr_aligned=0␛[0m    
D (227) boot: otadata[0]: sequence values 0xffffffff␛[0m
D (232) boot: otadata[1]: sequence values 0xffffffff␛[0m
D (237) boot: OTA sequence numbers both empty (all-0xFF) or partition table does not have bootable ota_apps (app_count=2)␛[0m
␛[0;32mI (248) boot: No factory image, trying OTA 0␛[0m
D (253) boot: Trying partition index 0 offs 0x10000 size 0x730000␛[0m
D (259) esp_image: reading image header @ 0x10000␛[0m
D (264) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)␛[0m
D (271) esp_image: image header: 0xe9 0x07 0x00 0x04 40081388␛[0m
␛[0;32mI (276) boot_comm: chip revision: 3, min. application chip revision: 0␛[0m
␛[0;32mI (284) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17984 ( 96644) map␛[0m
D (292) esp_image: free data page_count 0x00000032␛[0m
D (297) bootloader_flash: mmu set paddr=00010000 count=2 size=17984 src_addr=10020 src_addr_aligned=10000␛[0m
D (338) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)␛[0m
␛[0;32mI (338) esp_image: segment 1: paddr=0x000279ac vaddr=0x3ffbdb60 size=0x034a0 ( 13472) load␛[0m
D (343) esp_image: free data page_count 0x00000032␛[0m
D (348) bootloader_flash: mmu set paddr=00020000 count=1 size=34a0 src_addr=279ac src_addr_aligned=20000␛[0m
[... 496 lines cut ...]
host tx hci data; handle=0 length=9
ble_hs_hci_acl_tx(): 0x00 0x00 0x09 0x00 0x05 0x00 0x04 0x00 0x01 0x08 0x0c 0x00 0x0a
Number of Completed Packets: num_handles=1
handle:0 pkts:1
D (12135) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xdead to 0xbaad␛[0m
D (12135) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xbaad to 0xbaad␛[0m
D (12135) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xf00d to 0xf00d␛[0m
D (12145) NimBLECharacteristic: >> notify: length: 6␛[0m
D (12145) NimBLECharacteristic: << notify: No clients subscribed.␛[0m
D (14155) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xdead to 0xbaad␛[0m
D (14155) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xbaad to 0xbaad␛[0m
D (14155) NimBLEUUID: Comparing UUIDs; type 16 to 16; UUID 0xf00d to 0xf00d␛[0m
D (14165) NimBLECharacteristic: >> notify: length: 6␛[0m
D (14165) NimBLECharacteristic: << notify: No clients subscribed.␛[0m
Disconnection Complete: status=0 handle=0 reason=8
D (16155) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT␛[0m
Client disconnected - start advertising
D (16155) NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0␛[0m
␛[0;33mW (16165) NimBLEAdvertising: Advertising already active␛[0m
D (16165) NimBLEServerCallbacks: onDisconnect(): Default␛[0m
D (16175) NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0␛[0m
␛[0;33mW (16185) NimBLEAdvertising: Advertising already active␛[0m
```

</details>

h2zero, 2021-06-02:
That's very odd, the esp32 is also logging the disconnect as a timeout but I don't see anything to suggest why. Are you connected to it with a custom app? If so could you try with nRFconnect and check if you have the same error?

AlexHartford, 2021-06-03:
We have been using nRFConnect... We're pretty stumped :(

h2zero, 2021-06-03:
Ok, I will see if I can reproduce the issue here and hopefully find the cause. Which version/commit of IDF are you on?

AlexHartford, 2021-06-03:
We're on the latest platformio version (3.2.1) which I believe only uses esp-idf version 4.2.1.

h2zero, 2021-06-03:
Thanks, I don't use platformio so I'll have to look at the commit it's using and see what I can come up with.

AlexHartford, 2021-06-03:
https://github.com/AlexHartford/esp32-ble-test/blob/master/sdkconfig

h2zero, 2021-06-03:
Thanks, I should have noticed that there, focused on the code :smiley:

AlexHartford, 2021-06-04:
No worries! We have still been unable to track the issue down in the meantime :disappointed:

h2zero, 2021-06-05:
I haven't had time to test this yet, sorry. I should be able to tonight or tomorrow.

h2zero, 2021-06-05:
@AlexHartford I am unable to reproduce the issue. I used your repo and it connects/reconnects with no error.
I'll test this more later on with different client devices and see what happens.

AlexHartford, 2021-06-07:
Thanks for your help, when running it directly through idf it works, it appears platform.io was causing the issue somehow. We wouldn't have thought to try it if you had not been able to get it working, and you mentioned you don't use platform.io. Cheers!

closed 2021-06-07 by AlexHartford

h2zero, 2021-06-07:
Actually I tested it in PlatformIO and IDF and it worked the same in both. Maybe something is different in your setup but I don't know PlatformIO well enough to suggest anything.

Glad you have it working!