h2zero/esp-nimble-cpp #47: Cannot maintain stable connection

MatheusTomazella, 2021-06-20:
## Problem

### Background
I have a project which uses esp-NimBLE-cpp to handle bluetooth communication and it was working great. A few days ago, during the implementation of a non-related update on my codebase, every connection I tried to establish was immediately dropped. After a few hours trying to identify and fix the issue, I created a new minimal project that contained just the bluetooth module and the problem persisted, meaning my surrounding code was not causing it. 

### Environment

- Windows 10
- Esp32
- Vs code
- Esp-idf
- esp-nimble-cpp

### Expected
Maintain a stable connection.

### Actual
Connects and automatically disconnects.
The device no longer appears on the scan and it’s impossible to connect until the microcontroller is reset, and the cycle continues.

### Relevant correction attempts
- Creation of a new and minimal project
- Removal of the microcontroller from the circuit
- Perform tests using a different Android device

All failed

## Code

### main.cpp

```
#include <stdio.h>
#include "BleController.h"

static BleController   bleController;

extern "C" void app_main ( void ) {
    bleController = BleController( (char*) "device_test" );
}
```

### BleController.h
```

#pragma once

#include <string>
#include "NimBLEDevice.h"

class BleController {
    BLEServer*       server;
    BLEService*      service1;
    BLEService*      service2;
    BLEAdvertising*  advertising;

    BLECharacteristic* char1;
    BLECharacteristic* char2_1;
    BLECharacteristic* char2_2;
public:
    BleController ( std::string deviceName ) {
        BLEDevice::init( deviceName );

        server = BLEDevice::createServer( );

        service1 = server  ->createService( "2d0ec625-d550-4ed3-8068-def71c9aa7ee" );
        char1    = service1->createCharacteristic( "6e8e098a-b4ca-4b28-ba66-afb65f4c83e1" );

        service2 = server  ->createService( "662cfb42-f43c-469a-bedd-1b13b8d7ec28" );
        char2_1  = service2->createCharacteristic( "f0003b89-4239-40a8-bcf1-bd60699d6179" );
[... 579 lines cut ...]
ble_hs_hci_acl_tx(): 0x00 0x00 0x09 0x00 0x05 0x00 0x04 0x00 0x01 0x08 0x0a 0x00 0x0a 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=11 data=0x07 0x00 0x04 0x00 0x08 0x0a 0x00 0x0c 0x00 0x03 0x28 
rxed att command: read type req; conn=0 start_handle=0x000a end_handle=0x000c
txed att command: read type rsp; conn=0 length=21
host tx hci data; handle=0 length=27
ble_hs_hci_acl_tx(): 0x00 0x00 0x1b 0x00 0x17 0x00 0x04 0x00 0x09 0x15 0x0b 0x00 0x0a 0x0c 0x00 0xe1 0x83 0x4c 0x5f 0xb6 0xaf 0x66 0xba 0x28 
0x4b 0xca 0xb4 0x8a 0x09 0x8e 0x6e
Number of Completed Packets: num_handles=1
handle:0 pkts:1
Disconnection Complete: status=0 handle=0 reason=8
D NimBLEServer: ">> handleGapEvent: BLE_GAP_EVENT_DISCONNECT"
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15
0x06 0x20 0x0f 0x30 0x00 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x07 0x00
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x6 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
0x0a 0x20 0x01 0x01
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xa status=0
D NimBLEAdvertising: "<< Advertising start"
```

h2zero, 2021-06-20:
Hi @MatheusTomazella, this is a strange issue for sure if it was working fine before. Did you update IDF versions or make any changes in menuconfig? I can see the reason for disconnect appears to be a supervision timeout, what are you connecting to this server with?

MatheusTomazella, 2021-06-20:
I'm using nRF connect to test, but using the application developed for the system or the android configuration page to pair result in the same problem.
The IDF version is the same (1.0.3), and the only update regarding the environment was creating a new project to fix some cmake problems. The issue persists though project however, so I believe it is not the source.

I will update the issue and include the sdkconfig, I believe it could help.

DavidJRobertson, 2021-06-20:
The log shows you are using off 4.2.1, which has broken NimBLE. See
https://github.com/espressif/esp-idf/issues/7026

MatheusTomazella, 2021-06-20:
After updating the IDF to v4.2.2 the issue was resolved.
I assumed the vs code extension version was related to the IDF version and was not aware the second changed.
Thank you very much for your time, help and library.

closed 2021-06-20 by MatheusTomazella

h2zero, 2021-06-20:
Thanks for the info @DavidJRobertson. I have not tested with 4.2.1 but it looks like the fix is out with 4.2.2.