h2zero/esp-nimble-cpp #50: Compilation fails with idf 4.3/gcc 8.4

arjanmels, 2021-07-14:
A number of compile failures occurs (probably due to switch to gcc 8.4):
invalid conversion from 'const void*' to 'void*' [-fpermissive] in NMEBLEAdvertising
invalid conversion from 'int' to 'esp_log_level_t' [-fpermissive] in log_common.h

I worked around it by adding "idf_build_set_property(COMPILE_OPTIONS "-fpermissive" APPEND)" to the CMakeLists.txt

h2zero, 2021-07-14:
I do not get any compile errors with 4.3. Are you using the master branch of this repo?

arjanmels, 2021-07-14:
I am using 1.2.0 release.

The master branch have an error on freetos.cpp file missing.

Arjan

h2zero, 2021-07-14:
Yes, that has been corrected, please try with latest master commit.

arjanmels, 2021-07-14:
Great! 

Almost, I get:
'CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE' was not declared in this scope
'CONFIG_BTDM_SCAN_DUPL_TYPE' was not declared in this scope

Which are called:
CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE
CONFIG_BT_CTRL_SCAN_DUPL_TYPE
in my sdkconfig. I updated the nimconfig_rename.h file
`
#if defined(CONFIG_SCAN_DUPLICATE_TYPE) && !defined(CONFIG_BTDM_SCAN_DUPL_TYPE)
#define CONFIG_BTDM_SCAN_DUPL_TYPE CONFIG_SCAN_DUPLICATE_TYPE 
#endif

#if defined(CONFIG_BT_CTRL_SCAN_DUPL_TYPE) && !defined(CONFIG_BTDM_SCAN_DUPL_TYPE)
#define CONFIG_BTDM_SCAN_DUPL_TYPE CONFIG_BT_CTRL_SCAN_DUPL_TYPE 
#endif

#if defined(CONFIG_DUPLICATE_SCAN_CACHE_SIZE) && !defined(CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE)
#define CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE CONFIG_DUPLICATE_SCAN_CACHE_SIZE 
#endif

#if defined(CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE) && !defined(CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE)
#define CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE 
#endif

`
and now it works. 

Thanks a lot!

closed 2021-07-14 by arjanmels