h2zero/esp-nimble-cpp #52: Errors While Compiling for ESP32C3

x0054, 2021-07-17:
I am trying to compile the library for the `esp32c3` target. It compiles just fine for the `esp32` target, but when I switch targets to `esp32c3` I get errors. Here is one:

```
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../components/esp-nimble-cpp/src/NimBLEDevice.cpp:764:16: error: 'struct esp_bt_controller_config_t' has no member named 'mode'
         bt_cfg.mode = ESP_BT_MODE_BLE;
```

Best I can tell is what's happening is that during compile the wrong header file is used. For ESP32C3 we are supposed to use `components/bt/include/esp32c3/esp_bt.h`, instead it's grabbing  `components/bt/include/esp32/esp_bt.h`. I am not sure how to fix it or if it's an issue with IDF build system or the code. Still trying to wrap my head around it. Any help?

Ways to reproduce:

1. Create a new project.
2. Set Target to `esp32c3` (`idf.py set-target esp32c3`)
3. Compile, see errors :)

Feel free to close if this is NOT an a library issue, but rather an IDF issue.

h2zero, 2021-07-18:
Thank you for reporting this, it is a library issue. The C3 renamed that config variable to `bluetooth_mode` so a check will need to be added to use the correct variable for the C3.

h2zero, 2021-07-18:
@x0054 I have implemented the fix in #53, please check it out and let me know.

x0054, 2021-07-19:
@h2zero, Thank you very much, the fix worked. Everything compiles properly at the moment. Thank you for an awesome library!

closed 2021-07-19 by x0054

h2zero, 2021-07-19:
You're welcome :) thanks for testing.