h2zero/esp-nimble-cpp #56: Compilation issues on ESP-S3

BarakChamo, 2021-07-22:
Hi @h2zero 

I just got my hands on an S3 devkit and noticed compilation issues that seem similar to previous issues mentioning the C3.

In the case of the S3, the errors are:
`'struct esp_bt_controller_config_t' has no member named 'mode'`
`'struct esp_bt_controller_config_t' has no member named 'ble_max_conn'; did you mean 'ble_max_act'?`
`'struct esp_bt_controller_config_t' has no member named 'normal_adv_size'`
`'struct esp_bt_controller_config_t' has no member named 'scan_duplicate_type'`

I tried changing the `ifdef` check from `CONFIG_IDF_TARGET_ESP32C3` to `CONFIG_IDF_TARGET_ESP32S3` and that seems to have fixed the `mode` error, but wasn't sure what to do with the rest.

Also, I noticed you added a few more checks for `CONFIG_IDF_TARGET_ESP32`.
Are these limitations specific to `C3`?
If you wouldn't mind expanding on why these checks are in place and where the different BLE featuresets are documented that would be really helpful!

Thanks again for your hard work on this library 🙏

BarakChamo, 2021-07-22:
Another update:
seems to compile successfully with `esp-idf#master`, haven't gotten to the point of running it yet but will report back.

h2zero, 2021-07-22:
Hi @BarakChamo I have not tried compiling for the S3 yet. As you have pointed out there are differences that need to be accounted for on the different chips. Looks like you've made good progress.

The ESP32 checks are needed for some functions that are only available to it due to different controllers etc.. 

I would very much welcome a PR with you changes if you'd like to submit one.

BarakChamo, 2021-07-23:
Hey @h2zero 

So turns out installing IDF from master and switching the condition to target S3 solves all issues.

Very happy to submit a PR once 4.4 is released and we can see no other changes were added.

mickeyl, 2023-04-28:
@BarakChamo Could you close the issue then?

closed 2024-06-06 by h2zero