h2zero/esp-nimble-cpp #58: Secondary Services

justintconroy, 2021-09-07:
Is there any way to add a secondary service? It looks like I can only add primary services. I may need to reuse a secondary service as part of another primary service on the same device.

h2zero, 2021-09-07:
Hi, there is currently no support for that but it should be easy to add. I've never had a request for it or even seen this used anywhere before.

justintconroy, 2021-09-08:
A few of the standard BLE services say they can be or should be implemented as secondary services.
* [Audio Input Control Service](https://www.bluetooth.com/specifications/specs/audio-input-control-service-1-0/)
* [Object Transfer Service](https://www.bluetooth.com/specifications/specs/object-transfer-service-1-0/)
* [Volume Offset Control Service](https://www.bluetooth.com/specifications/specs/volume-offset-control-service-1-0/)

are a few of the one's I've encountered. Right now, I'm specifically looking at Object Transfer Service. I've implemented a scaled down version of that (Connection Oriented Channels aren't supported on the ESP32 as far as I can tell, so I had to improvise a bit). I have another, distinct set of objects that I might want to throw this on top of. In that case, I want both services to be secondary services included in separate primary services.

h2zero, 2021-09-08:
Thanks for the info, I guess bluetooth had to have a reason for secondary services somewhere and I haven't seen those before.
As a user, how would you like to see the interface for this work?

> Connection Oriented Channels aren't supported on the ESP32 as far as I can tell

I believe they are, there are options in menuconfig for them (for NimBLE I thought I saw them at least).

justintconroy, 2021-09-08:
There's a config for "max number of CoCs", but I couldn't find the APIs to actually access them. Nor could I find any examples. Maybe I just didn't look hard enough...

justintconroy, 2021-09-08:
Sorry, I didn't actually answer your question. I think maybe just something like `NimBLEService::addService(...)` or `NimBLEService::createService(...)` similar to how characteristics are added to a service.

h2zero, 2021-09-08:
Thanks, I'll have a look at adding this. NimBLE internals might dictate a different interface but I'll see what I can do.

h2zero, 2021-09-12:
I've created a PR #59, If you could test it an provide feedback it would be greatly appreciated 😄

mickeyl, 2023-04-28:
@justintconroy Any feedback on this yet?