h2zero/esp-nimble-cpp #60: References to arduino-esp32

lgLindstrom, 2021-09-26:
Hi

I am running a essp-idf project with some arduino legacy code. At the moment I would prefer not rewrite those parts so I have installed arduino-esp32 as a component in my project.

To be able to use BLE nimble stack I installer esp-nimble-cpp. When building I have some error I don't understand like :

`../idf_wrappers/esp-nimble/src/NimBLEDevice.cpp:46:10: fatal error: esp32-hal-bt.h: No such file or directory`

esp32-hal-bt.h is part of the arduino-esp32 library,,,  should there be references to that in esp-nimble-cpp???

h2zero, 2021-09-26:
If you are using arduino as an idf component as well as this library it needs to include that file so that the arduino code won't release the BLE memory. That's the only reason for including that.

lgLindstrom, 2021-09-26:
Okay. Thanks for the explanation.

JakubRybakowski, 2021-11-22:
```
../components/arduino/libraries/BluetoothSerial/src/BTAddress.h:14:10: fatal error: esp_gap_bt_api.h: No such file or directory
 #include <esp_gap_bt_api.h> // ESP32 BT
```

I have a similar problem.

After changing to nimble. Arduino as component won't compile. I disabled BluetoothSerial in sdkconfig but still the same.

h2zero, 2021-11-22:
Looks like Bluetooth is not enabled in your config? Can you share the IDF and arduino core commit ID?

JakubRybakowski, 2021-11-23:
Bluetooth is on. Host is nimBLE

If I comment these lines in the arduino component, everything works:

components/arduino/CMakeList.txt

```
  #libraries/BluetoothSerial/src/BluetoothSerial.cpp
  #libraries/BluetoothSerial/src/BTAddress.cpp
  #libraries/BluetoothSerial/src/BTAdvertisedDeviceSet.cpp
  #libraries/BluetoothSerial/src/BTScanResultsSet.cpp
```

But it is not an elegant solution.

Arduino: 399f4ecbb3a4cef21e2bffa37adb6190356dfb76 (https://github.com/espressif/arduino-esp32/commit/399f4ecbb3a4cef21e2bffa37adb6190356dfb76)
idf v4.4 35b20cadce65ce79c14cf2018efc87c44d71ab21 (https://github.com/espressif/esp-idf/commit/35b20cadce65ce79c14cf2018efc87c44d71ab21)

h2zero, 2021-11-23:
Thanks!

Looking through the arduino core I can see the header only checks `CONFIG_BT_ENABLED` and should be `CONFIG_BT_BLUEDROID_ENABLED`

Obviously, that would require a PR to fix. Potentially you could define the include guard macro in your project Cmake to avoid altering the component files.

JakubRybakowski, 2021-11-23:
Thank you, this looks like a better solution.

Thanks for your work and this repository.

h2zero, 2021-11-23:
You're very welcome, let me know if that works. In the meantime I will propose the necessary PR to arduino-esp32.

JakubRybakowski, 2021-11-23:
Yes it works.
Here PR:
https://github.com/espressif/arduino-esp32/pull/5920

h2zero, 2021-11-23:
Awesome, thanks!

ToyboxZach, 2023-02-14:
Is there a way to fix the original issue?

I am having that compiling issue as well and don't know how to make the component be able to find the include file

travisjayday, 2023-09-18:
I had this issue using Platform io with 
`framework = arduino, espidf` in platformio.ini

To solve it, I removed 'arduino' from the framework field, and installed arduino as a separate component.

closed 2024-06-06 by h2zero