h2zero/esp-nimble-cpp #62: Question:  Gatt error 133 for basic examples ble_client and ble_notify

goofy2k, 2021-11-14:
Hi, thanks for your excellent content here!  I am new to BLE and Nimble, so I work through your docs and examples. I have an application where I want to push data from one device to another, so I started to get ble_client and ble_notify running under esp-idf. In the end I succeeded to have both apps running and I see the data coming in on the ble_client device !

I used the nRF Connect app to get this far. I want to set my next steps with building my application around both examples. I expect that using the nRF Connect app will be very valuable in this,  but now I noticed that the app does not connect to the ble_notify device when both peers are running. Does this have something to do with a limitation in the number of connections? Or is the ble_notify example simply not designed to serve two clients?

h2zero, 2021-11-16:
The BLE_Notify example does not have support for multiple client connections. When a connection occurs, advertising is stopped and needs to be manually started. It would be simple enough to add this by making a call to `BLEDevice::startAdvertising();` in the `onConnect()` callback.

goofy2k, 2021-11-16:
Thanks for your answer and suggestion !  Learning a lot :-)

closed 2021-11-16 by goofy2k