h2zero/esp-nimble-cpp #66: Uable to run basic ble scan example. ESP32 crash

mb-software, 2022-01-07:
Hello,

I'm new to ESP-IDF and trying to do my first steps with nimble.
I tried to get the basic ble scan example running, but failed. I'm using the exact same code from the example. I did activate NimBLE in menuconfig, keeping all further NimBLE settings with their defaults. (I thought the defaults should work for this basic test example).

But obviously I was wrong. The project compiles and uploads fine, but the esp32 (wemos mini 32) immediatelly goes to bootloop with crash: 
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`

This also happens when I comment out everything except the BLEDevice::init("") call, so the BLEDevice::init() seems to be causing the issue.

Is there anything else I need to change in the settings to get this simple example running?
Any help for a newbie appreciated :)

h2zero, 2022-01-08:
Hi @mb-software what version of IDF are you using?

mb-software, 2022-01-08:
Hi, thanks for your support. I'm using IDF 4.3.1 ( default when creating a new esp32 IDF project in platformio)

h2zero, 2022-01-08:
@mb-software I have not been able to reproduce this using IDF 4.3.1 directly (not in platformio). Could you provide some log output and decode the backtrace if possible?

h2zero, 2022-01-08:
I just tested in platformio as well and everything looks fine there. Maybe something wrong with your config?

mb-software, 2022-01-09:
Honestly I don't know where to start looking for a "wrong" configuration. Like I said, I only let platformio create a project with default configuration for my esp32 board and the only change was enabling Nimble ble stack.
I will try to start with a new project once again and give it another try. If I get the same result maybe I can somehow share the config with you...

h2zero, 2022-01-09:
Here is my config file, you may need to change the extension to match your board. I also recommend deleting the .pio folder before using this.

[sdkconfig.esp32dev.txt](https://github.com/h2zero/esp-nimble-cpp/files/7835068/sdkconfig.esp32dev.txt)

mb-software, 2022-01-09:
Hi, thanks so much for your help. It's working now. I think simply deleting the .pio is what did the trick in the end... something must have been messed up there...

closed 2022-01-09 by mb-software

h2zero, 2022-01-09:
Great, thanks for the update.