h2zero/esp-nimble-cpp #68: Help with security in Basic examples?

iboguslavsky, 2022-01-19:
I'm using stock BLE_server and BLE_client examples, but trying to add basic security to it. The code appears to ignore security altogether, and happily reads / writes characteristic's values (which are set to be encrypted, see immediately below):

On the server:
```
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                        CHARACTERISTIC_UUID,
                                        NIMBLE_PROPERTY::READ |  NIMBLE_PROPERTY::READ_ENC |
                                        NIMBLE_PROPERTY::WRITE |  NIMBLE_PROPERTY::WRITE_ENC);
```
```
  BLEDevice::init("Long name works now");

  // Add security
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_DISPLAY);
  NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_MITM);
  NimBLEDevice::setSecurityPasskey(123456);
```

On the client:
```
  BLEDevice::init("");

  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_DISPLAY);
  NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_MITM);
```

What am I doing wrong / missing?

iboguslavsky, 2022-01-20:
Adding corresponding ClientCallback / ServerCallback classes fixes the problem.

closed 2022-01-20 by iboguslavsky

h2zero, 2022-01-20:
Sorry for the delay getting back to you. I was going to do some testing on the security stuff soon and was going to try your setup to see what was happening.

Yes, without the callbacks it would use the defaults which will happily connect with the 123456 pin.