h2zero/esp-nimble-cpp #70: IDF + ARDUINO build issue

Koxx3, 2022-01-27:
Hello,
I used your very good lib with arduino only before, no trouble.

Now, I am migrating to IDF + ARDUINO with platformio.

It doesn't build.

platformio.ini
```
[env]
platform = espressif32
framework = arduino, espidf
board = esp32dev
platform_packages =
  framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0

build_flags =
	-DESP32=1
	-DARDUINO_ARCH_ESP32=1
	-DARDUINO=100
	-Wformat-contains-nul
	-Wno-format-overflow
	-Wno-error=format-truncation

```

lib is placed in "components" folder :
[image]

sdkconfig
```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
# CONFIG_BT_NIMBLE_ENABLED is not set
# CONFIG_BT_CONTROLLER_ONLY is not set
```

build result
```
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.d011dd7
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
...
Building in release mode
Generating assembly for .pio\build\smartcontroller_minimo\settingspage.html.gz.S
Compiling .pio\build\smartcontroller_minimo\src\BLE\BluetoothHandler.o
Compiling .pio\build\smartcontroller_minimo\src\Battery\Battery.o
In file included from components/esp-nimble-cpp/src/NimBLEBeacon.h:18,
                 from src/BLE/BluetoothHandler.cpp:1:
components/esp-nimble-cpp/src/NimBLEUUID.h:22:10: fatal error: host/ble_uuid.h: No such file or directory
 #include "host/ble_uuid.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
...
...
In file included from components/esp-nimble-cpp/src/NimBLEDevice.h:37,
                 from src/BLE/BluetoothHandler.h:11,
                 from src/Buttons/Buttons.h:9,
                 from src/Buttons/Buttons.cpp:1:
components/esp-nimble-cpp/src/NimBLEUtils.h:16:10: fatal error: host/ble_gap.h: No such file or directory
 #include "host/ble_gap.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
...

any idea how I could solve this ?

h2zero, 2022-01-27:
Please try enabling nimble in your config or adding my esp-nimble-component library to your components folder.

Koxx3, 2022-01-29:
not sure I understand what you mean.

in sdkconfig, If I try to enable both (bluedroid & nimble), it automatically reset with only one of them.

and for the esp-nimble-component library; it is alread on the component folder as visible here :
[image]

did I do something wrong ?

flynny75, 2022-10-28:
Hello,

I am in a similar situation, trying to use nimble with both arduino and espidf frameworks. I am encountering a different error: `components/esp-nimble-cpp-1.4.0/src/NimBLEDevice.cpp:52:12: fatal error: esp32-hal-bt.h: No such file or directory`

I have attached a skeleton project that causes the issue.
[NimbleBuild.zip](https://github.com/h2zero/esp-nimble-cpp/files/9892275/NimbleBuild.zip)

flynny75, 2022-10-29:
This seems to be a problem with the CMakeLists when using both espidf and arduino frameworks.

Removing the condition `if("arduino" IN_LIST BUILD_COMPONENTS OR __hack_component_targets MATCHES "__idf_arduino")` and appending `framework-arduinoespressif32` rather than `arduino` seems to resolve the issue

h2zero, 2024-06-14:
Fixed in #122

closed 2024-06-14 by h2zero