h2zero/esp-nimble-cpp #71: arduino nimble coded phy esp32

iamcalledusman, 2022-03-27:
Hi All!

Is there any option to enable the **Coded PHY** in Arduino-Nimble? If yes can you please me a guide for that? 
Preferably in the client-server app or beacon scanner app.

Thanks, in advance!

h2zero, 2022-03-27:
Only coded phy scanning is currently available to this repo. That support has not been transferred to the arduino port yet. If you want to use this repo with IDF and arduino as a component it should work.

iamcalledusman, 2022-03-28:
Thank you so much! for the response. Can you please guide me for the coded PHY advertisement with nimble for any other repo? 

I will use this for the coded PHY scanning and I need a coded phy advertisement.

iamcalledusman, 2022-03-28:
Again thank you for the response!

Can u please guide me how to use coded phy repo for esp-idf ? 

please if you can mention the coded phy repo to use with esp-idf (Nimble host ).

Thanks in advance !

h2zero, 2022-03-28:
You just need to enable it in menuconfig under the NimBLE host options.

h2zero, 2022-04-03:
@iamcalledusman The initial implementation is [here](https://github.com/h2zero/esp-nimble-cpp/tree/extended-advertising).

Sorry I don't have any examples prepared, you will have to look through the comments in the code and the doc's added to the commit in that branch. More to be done soon, but if you get it working I'd love to hear your feedback.

closed 2022-04-10 by h2zero