h2zero/esp-nimble-cpp #83: L2CAP channel?

mickeyl, 2022-05-27:
First off, thanks for this great set of classes!

I'm using an esp32c3 and need to transfer a huge stream of data to iPhone (and Android) over BLE and wonder whether your classes allow me to create an L2CAP channel directly? That way my hope is I would not have to implement manual fragmentation and assembly of data using individual characteristics.

h2zero, 2022-05-28:
Hi @mickeyl, you're quite welcome.

There is no support for l2cap in this library at this time. Always happy to accept a PR though 😃

mickeyl, 2022-05-29:
Alright. Will do as soon as I get it working on the C level… right now I'm [still blocked](https://github.com/espressif/esp-idf/issues/9044).

mickeyl, 2022-06-05:
FYI: I have started to get some things working on C level. I will create a bunch of abstractions that hopefully match your style. Bear with me though, as my last encounter with C++ stems from using Visual C++ with Windows NT 3.5.1 (in the late 1990s, if I recall correctly), my C++ is a bit rusty.

I'm developing the code [here](https://github.com/mickeyl/esp-nimble-cpp/) and will open a pull request once I have something workable.

h2zero, 2022-06-06:
Awesome, looking forward to it!

mickeyl, 2023-04-21:
Quick heads-up: I made huge progress and will open a pull request for review next week.

mickeyl, 2023-04-28:
https://github.com/h2zero/esp-nimble-cpp/pull/116

mickeyl, 2025-04-29:
https://github.com/h2zero/esp-nimble-cpp/pull/324 has been merged, so we can close this issue for now.

closed 2025-04-29 by mickeyl