h2zero/esp-nimble-cpp #88: question - update from 1.2 to 1.3.3

mitchjs, 2022-06-30:
Hi there, 
been a while for me, but i needed to make an update to my existing project
so i figured id update the Nimble-CPP component
i had to make 1 change to my code

```
// create main service
pService = pServer->createService(NimBLEUUID(SERVICE_UUID), 75);
to
// create main service
pService = pServer->createService(NimBLEUUID(SERVICE_UUID));
```

im just wondering is this automatic now to allocate the "space" for all the characteristics?

thanks again for great work on this
mitch

mitchjs, 2022-06-30:
oh i saw on some githubs, a discussion section you can add, for talking about things like this

mitch

h2zero, 2022-06-30:
Hi @mitchjs, been a while for sure 😄. That parameter was never used by this library so I just removed it. Minor API breakage I figured.

mitchjs, 2022-06-30:
i thought it was, if the value was too small, characteristics wouldnt get made, it was for some memory allocation
(i cant test my new code, yet... i updated whole project to the ESP32-S3 and waiting on dev board to test out)
been a while since i needed to look at code, so what do i know :)

closed 2024-06-14 by h2zero