h2zero/esp-nimble-cpp #92: getValue() not returning correct value in onWrite callback

tobobo, 2022-07-18:
Hi, thank you for your work on this library! It's saving me tons of time converting my esp-idf project from Bluedroid to NimBLE.

I've noticed that when I write to a characteristic, the value written seems to get corrupted in between when it is written and when I read it in `onWrite`. 

In `NimBLECharacteristic`, I've added logging which confirms the correct value is set and can be read immediately after the call to `setValue`

```
            ESP_LOGI("NimBLE", "Setting characteristic value");
            ESP_LOG_BUFFER_HEXDUMP("NimBLE", buf, len, ESP_LOG_INFO);
            pCharacteristic->setValue(buf, len);
            ESP_LOGI("NimBLE", "After set value");
            ESP_LOG_BUFFER_HEXDUMP("NimBLE", pCharacteristic->getValue().data(), pCharacteristic->getValue().length(), ESP_LOG_INFO);
            pCharacteristic->m_pCallbacks->onWrite(pCharacteristic);
            pCharacteristic->m_pCallbacks->onWrite(pCharacteristic, &desc);
```

This logs the following:

```
I (51469) NimBLE: Setting characteristic value
I (51479) NimBLE: 0x3ffec010   00 14 00 7b 22 73 73 69  64 22 3a 22 57 61 74 65  |...{"ssid":"Wate|
I (51489) NimBLE: 0x3ffec020   72 54 6f 77 65 72 22                              |rTower"|
I (51499) NimBLE: After set value
I (51499) NimBLE: 0x3ffecec0   00 14 00 7b 22 73 73 69  64 22 3a 22 57 61 74 65  |...{"ssid":"Wate|
I (51509) NimBLE: 0x3ffeced0   72 54 6f 77 65 72 22                              |rTower"|
```

However, in my `onWrite` callback, I'm logging the value from the characteristic and seeing a corrupted value.

```
  const uint8_t *writeValue = (uint8_t *)(characteristic->getValue().c_str());
  const size_t writeLength = characteristic->getValue().length();

  ESP_LOGI(TAG, "Long write segment, %i bytes", writeLength);
  ESP_LOG_BUFFER_HEXDUMP(TAG, writeValue, writeLength, ESP_LOG_INFO);
```

This logs the following

```
I (51529) LongWriteHandler: Long write segment, 23 bytes
I (51539) LongWriteHandler: 0x3fff443c   64 43 fe 3f 64 43 fe 3f  64 22 3a 22 57 61 74 65  |dC.?dC.?d":"Wate|
I (51549) LongWriteHandler: 0x3fff444c   72 54 6f 77 34 44 ff                              |rTow4D.|
```

Any insight in to how to access the uncorrupted value from my onWrite handler is appreciated! Thank you!

tobobo, 2022-07-20:
Thank you! It was indeed fixed by changing my code in the callback to

```
  std::string writeValueString = characteristic->getValue();
  const uint8_t *writeValue = (uint8_t *)(writeValueString.c_str());
  const size_t writeLength = writeValueString.length();
```

It's a bit counterintuitive to me because the value is ascii data with a binary header, but this works fine for my purposes. I'm also new to C/C++ though and have used std::string very little so there might be some conventions here I'm missing. Thank you again for your help and for making a great library!

h2zero, 2022-07-20:
You're very welcome!

>It's a bit counterintuitive to me because the value is ascii data with a binary header, but this works fine for my purposes. I'm also new to C/C++ though and have used std::string very little so there might be some conventions here I'm missing.

Yes it's a bit strange to use the std::string library in this manner, it's just the way the original library handled data, so it was kept that way here. That said, it was quite clever since std::string is more or less just an extension of std::vector<uint8_t>.

So your `onWrite` handler could also be this:
```
auto val =  characteristic->getValue();
ESP_LOGI(TAG, "Long write segment, %u bytes", val.size());
ESP_LOG_BUFFER_HEXDUMP(TAG, &val[0], val.size(), ESP_LOG_INFO);
```

tobobo, 2022-07-20:
ah, interesting—I'll have to read up on the `auto` keyword. Thanks again!

closed 2022-07-27 by tobobo