h2zero/esp-nimble-cpp #98: Difference in default BLE and nimBLE when having  multiple advertisements

AndrewMagpie, 2022-09-06:
When using the default BLE library and advertising a 128 bit followed by a 16 bit advertisement I  see both in nRFConnect.
When using the nimBLE library I only see the the 128 bit advertisement (or whatever was added first with `addServiceUUID()`).

Default BLE library:
[image]

nimBLE:
[image]

```
    #ifdef DO_NIMBLE
    std::string strBLEName = "nimBLE Test";
    #else
    std::string strBLEName = "BLE Test";
    #endif

    //Init
    BLEDevice::init(strBLEName.c_str());

    BLEServer*  pServer = BLEDevice::createServer();

    #ifdef DO_NIMBLE
    uint32_t uProperties = NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE;
    #else
    uint32_t uProperties = BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE;
    #endif

    BLEService * pService128 = nullptr;
    {
      #define SERVICE_UUID_128 "ca65f688-5a42-4e4a-b19f-e59e70758573"
      #define CHARACT_UUID_128 "d7afc818-eab0-4363-b08c-fa39d0989583"

      pService128 = pServer->createService(SERVICE_UUID_128);
      BLECharacteristic* pChar = pService128->createCharacteristic(CHARACT_UUID_128, uProperties);
      pChar->setValue(std::string("UUID = 128"));
      pService128->start();
    }

    BLEService * pService16 = nullptr;
    {
      #define SERVICE_UUID_16 ((uint16_t)0x180A) /* Device info */
      #define CHARACT_UUID_16 ((uint16_t)0x2A00) /* Device name */

      pService16 = pServer->createService(SERVICE_UUID_16);
      BLECharacteristic* pChar = pService16->createCharacteristic(CHARACT_UUID_16, uProperties);
      pChar->setValue(std::string("UUID = 16"));
      pService16->start();
    }

    BLEAdvertising* pAdvertising = BLEDevice::getAdvertising();
    pAdvertising->setScanResponse(true);
    pAdvertising->setMinPreferred(0x06);
    pAdvertising->setMaxPreferred(0x12);
    pAdvertising->setAppearance(0x0140);  //Appearance=Display

    //For nimBLE, order determines which is actually advertised.
    pAdvertising->addServiceUUID(pService128->getUUID());
    pAdvertising->addServiceUUID(pService16->getUUID());

    #ifdef DO_NIMBLE
    bIsAdvertising = pAdvertising->start();
    #else
    pAdvertising->start();
    bIsAdvertising = true;
    #endif
```
How do I get nimBLE to advertise both UUIDs?
Using PlatformIO on a ESP32S3.

h2zero, 2022-09-07:
Hi @AndrewMagpie, I believe the cause to be in the different handling of the AD flags in the data count, I still need to verify this. For an explanation, the AD flags, IIRC, are accounted for differently in the advertisement data size with the original library, and in your situation you have now exceeded the data limit by 3 bytes. For now, if you don't need it I would suggest removing  these calls:
```
    pAdvertising->setMinPreferred(0x06);
    pAdvertising->setMaxPreferred(0x12);
```
This should be sufficient to resolve the issue and I will look into better handling of this.

AndrewMagpie, 2022-09-09:
@h2zero removing those fixed the advertisement but were there to get around an issue we were having with iPhones.

AndrewMagpie, 2022-09-09:
Please let me know if/when there is a fix for this so that we can start using those two APIs again.

h2zero, 2022-09-09:
I don't think that issue exists any longer, I have not yet heard of it being a problem since I started this repo and that functionality was removed initially.

closed 2024-06-08 by h2zero