h2zero/esp-nimble-cpp #104: getCharacteristic() makes ESP32 crash in arduino IDE

Patricksmtvac, 2022-10-06:
Hi @h2zero. In my project the ESP32 is connected to 2 BLE temperature beacons. The code runs a while and then crashes. Using ExceptionDecoder, it says that NimBLERemoteService.cpp line 96 causes this crash. Should i make any changes in this file? I'm using Arduino IDE v1.8.19. 
thank you
Patrick
[image]

h2zero, 2022-10-06:
Looks like you are calling the function on a nullptr. Try checking if the service/characteristic is null first.

Patricksmtvac, 2022-10-07:
I'm new to programming and ESP32 stuff. I really don't understand why in the NimBLE_Client example are these lines

       `NimBLERemoteService* pSvc = nullptr;
        NimBLERemoteCharacteristic* pChr = nullptr;
        NimBLERemoteDescriptor* pDsc = nullptr;` 

why assign the pointers to nullptr??

Anyways, here is a  little snippet of the code i'm using. I'm really stuck in here. :(

`bool connectToServer() 
{
    NimBLEClient* pClient = nullptr;

    /** Percorre a lista de sensores definida no Array de MACs **/
    for (int i=0; i<sizeof Addr/sizeof Addr[0]; i++)
    {  
      /** Check if we have a client we should reuse first **/
      
      if(NimBLEDevice::getClientListSize()) 
      {
         /** Special case when we already know this device, we send false as the
         *  second argument in connect() to prevent refreshing the service database.
         *  This saves considerable time and power. */
         
        pClient = NimBLEDevice::getClientByPeerAddress(NimBLEAddress(Addr[i][0]));
  
        if(pClient)
        {
            if(!pClient->connect(false)) 
            {
                NimBLEDevice::deleteClient(pClient); 
                Serial.println("     - Reconexão falhou...");
            }
            else
              Serial.println("     - Reconectado ao Sensor...");
        }
        /** We don't already have a client that knows this device,
         *  we will check for a client that is disconnected that we can use.
         */
        else 
            pClient = NimBLEDevice::getDisconnectedClient();
      }

      /** No client to reuse? Create a new one. */
      if(!pClient) 
      {
      
        if(NimBLEDevice::getClientListSize() >= NIMBLE_MAX_CONNECTIONS) 
            Serial.println("     - Número máximo de Sensores conectados! Impossível Conectar"); 
            
        pClient = NimBLEDevice::createClient(NimBLEAddress(Addr[i][0]));
        Serial.println("     - Novo Sensor foi criado...");

        pClient->setClientCallbacks(&clientCB, false);
        /** Set initial connection parameters: These settings are 15ms interval, 0 latency, 120ms timout.
         *  These settings are safe for 3 clients to connect reliably, can go faster if you have less
         *  connections. Timeout should be a multiple of the interval, minimum is 100ms.
         *  Min interval: 12 * 1.25ms = 15, Max interval: 12 * 1.25ms = 15, 0 latency, 51 * 10ms = 510ms timeout
         */
        pClient->setConnectionParams(12,12,0,51);

        /** Set how long we are willing to wait for the connection to complete (seconds), default is 30. */
        pClient->setConnectTimeout(10);    //5    

        if (!pClient->connect(false)) 
        {
            /** Created a client but failed to connect, don't need to keep it as it has no data */
           //NimBLEDevice::deleteClient(pClient);
           Serial.println("     - Conexão falhou, vai apagar o Sensor...");
        };
      }
     
                       
      /** Se conseguiu conectar, então exibe os resultados **/
      if (pClient->isConnected())
      {     
        Serial.println("===========================================");
        Serial.print("1. Conectado em: ");
        Serial.println(Addr[i][1]);
//        Serial.print("2. MAC.........: ");
//        Serial.println(pClient->getPeerAddress().toString().c_str());
//        Serial.print("3. RSSI........: ");
//        Serial.print(pClient->getRssi());
  
        /** Now we can read/write/subscribe the charateristics of the services we are interested in */
        NimBLERemoteService* pSvc = nullptr;
        NimBLERemoteCharacteristic* pChr = nullptr;
        NimBLERemoteDescriptor* pDsc = nullptr;

        
        
        pSvc = pClient->getService("aa20"); 
        pChr = pSvc->getCharacteristic("0xaa21");// Exception Decoder Indicates that here is the problem

        String newValue = "Canopus: " + String(millis()/1000);
        
        pChr->writeValue(newValue.c_str(), newValue.length());    
        std::string value = pChr->readValue();

        float temp = value[1]*100+value[2];
        if(value[1]==1)
          temp=temp*-1;

           float tempDef = temp/100;
           
              if(Addr[i][0] == addr0)
                    {
                       TempRet = tempDef;
                      Serial.printf("ret: ");
                      Serial.println(TempRet);
                    }
                    else if (Addr[i][0] == addr1)
                    {
                         TempIns = tempDef;
                        Serial.printf("ins: ");
                         Serial.println(TempIns); 
                    }//else if

         
      }   // if client connect
    } //Loop For
    
    return true;
}`

h2zero, 2022-10-08:
> why assign the pointers to nullptr??

This to ensure they have a known state before use, allowing us to check them for the value of `nullptr`.

Which is what you need to do here:
```
    pSvc = pClient->getService("aa20"); 
    pChr = pSvc->getCharacteristic("0xaa21");// Exception Decoder Indicates that here is the problem
```

If your call to `pSvc = pClient->getService("aa20"); ` was not successful, then `pSvc` is a `nullptr`, which causes the exception on the next line. So here is what you need to do:
```
    pSvc = pClient->getService("aa20"); 
    if (pSvc == nullptr) {
       // Handle error
    }
    pChr = pSvc->getCharacteristic("0xaa21"); // Suggest removing the "0x" here
    if (pChr == nullptr) {
       // Handle error
    }
```

Patricksmtvac, 2022-10-10:
It worked pretty well, thank you so much for your help :)

closed 2022-10-10 by Patricksmtvac