h2zero/esp-nimble-cpp #105: Library doesnt work properly (Something to do with mem release probably)

Baby1991, 2022-10-16:
Hello!
I have recently migrated from Arduino NimBLE library and have succeded in porting the code and it worked. But today after making some unrelated changes in code i tested if BLE still works, and its weird. I starts a server but gives me a warning: 
`W (1655) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK` 
and the server shows up on my BLE scanner. When i try to connect to it, the server just forcibly disconnects me with error:
`I (190418) NimBLE: GAP procedure initiated: advertise; `
`I (190424) NimBLE: disc_mode=2`
`I (190428) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0`
`I (190438) NimBLE: `

`I (190986) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19`
I am really unsure why this is happening as i havent touched BLE code.

closed 2022-10-18 by Baby1991

h2zero, 2022-10-18:
Sorry I forgot to get to this. Did you resolve the problem?

Baby1991, 2022-10-26:
@h2zero Eh not really, but i managed to go arround the problem by disabling nvs bonds. Now it works like a charm!

Baby1991, 2022-11-12:
@h2zero Sorry for the long delay in responding, honestly no idea, may be something filling up nvm or something corrupted. Worls like for a few times then starts working worse and worse, im not sure why in the slightest... Sorry for not being of any help