h2zero/esp-nimble-cpp #121: how to peripheral get the services of the central

arif-ozcan, 2023-05-28:
my use scenario is:

1- start advertising
2- connect to advertising device through nRF Connect from mobile
3- get custom client services of the central device (mobile)

I cannot access the services of the central devices from my advertising-peripheral device. A Client instance always wants to connect to the advertised device.

Please let me know, Am i missing any point ? If not, is there any example to cover my scenario ? 

Thanks in Advance !

h2zero, 2023-05-28:
Hello, I can't think of any way to do this currently. It could be done with some significant changes to the code, I do not have time for doing that however. If you'd like to explore this and submit a PR I'd be more than happy to accept it.

arif-ozcan, 2023-05-29:
Hi, thanks for the quick reply. 

Actually, i have started some modification on the code yesterday and could get service and characteristics by uuid. I hope i  submit a PR on next days after a few test.

```
int NimBLEServer::handleGapEvent(struct ble_gap_event *event, void *arg)
{
    NIMBLE_LOGD(LOG_TAG, ">> handleGapEvent: %s", NimBLEUtils::gapEventToString(event->type));

    int rc = 0;
    NimBLEConnInfo peerInfo;
    NimBLEServer *pServer = NimBLEDevice::getServer();
    NimBLEClient *pClient = (NimBLEClient *)arg; // arg pointer is not used before in this callback. 
...

...

pClient->setConnID(event->connect.conn_handle); // should be set connID for retrieveServices()
pServer->m_pServerCallbacks->onConnect(pServer, pClient, peerInfo); // a pointer of client instance in addition to server instance
}
```

h2zero, 2023-05-30:
Interesting approach, glad to see you're making progress. Happy to review when you're ready.

finger563, 2024-03-29:
@arif-ozcan Did you end up making a fork with the changes needed for this?