h2zero/esp-nimble-cpp #127: Hello BLE NimBLE project will not build when C but only C++ / VSC Intellisense

MortenPedersenDK, 2023-08-05:
Hi

I am new to ESP-IDF and BLE and trying to get a "Hello BLE" project running with NimBLE:

```
#include <stdio.h>
#include "NimBLEDevice.h"

void app_main(void)
{
    NimBLEDevice::init("NimBLE");
    
    NimBLEServer *pServer = NimBLEDevice::createServer();
    NimBLEService *pService = pServer->createService("ABCD");
}
```

I created a new empty project, added a components folder and unzipped "esp-nimble-cpp-master" to it.

I face multiple issues:

### Compile error in NimBLEAddress.h line 31:

```
#include <string>

fatal error: string: No such file or directory

```

Switching to C++ by renaming main.c to main.cpp and adding extern "C" to main method fixed this, and project now compiles.

However - as I see the samples, they use C files and I would like to understand why main.c does not work in my setup?

### Intellisense shows lots of errors:

For example:
[image]

and

[image]

When I was having the main.c file it was the :: in NimBLEDevice::init it did not like.

My c_cpp_properties.json is pretty standard:

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "/Users/morten/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
```

I guess there is some configuration I am missing, but cannot seem to find out what/where.

### Environment

- esp-nimble-cpp as of aug 4th 23
- ESP-IDF v5.1
- VS Code 1.78
- MacOS Ventura 13.4.1 (c)
- Python 3.9.6

### Project structure
[image]

Any advice appreciated - thanks!

MortenPedersenDK, 2023-08-05:
OK, I just saw one of the examples in this repo, and that it also uses C++. I must have been tired when I ran in to that problem, as it all makes sense that it must be C++ and not C. So please disregard that. If anyone has input on the intellisense problem, I will gladly hear from you. Thanks!

h2zero, 2023-08-12:
Try adding the line `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` to your c_cpp_properties.json.

MortenPedersenDK, 2023-08-12:
Yes - that fixed it! Thank for helping out! :-)

closed 2023-08-12 by MortenPedersenDK