h2zero/esp-nimble-cpp #133: Callbacks never called

jakubarm, 2023-10-06:
Hello,

I have upgraded to IDF 5.1 and esp-nimble-cpp to the latest master.
The BL peripheral device is visible and can be operated using mobile client app (e.g. nRF Client) but any custom callbacks (characteristic read/write, connect/disconnect) are not called. Esp-nimble-cpp provided no info log when activated in menuconfig nor nimble debug showed nothing (only advertising events). There is probably some change or I am doing something wrong using the new version. My init function:

```
NimBLEDevice::init(ss_name.str());
//BLEDevice::setPower(ESP_PWR_LVL_P9);
NimBLEDevice::setMTU(517);//default should be 500
server = NimBLEDevice::createServer();
server->setCallbacks(new ServerCustomCallbacks());
advertising = NimBLEDevice::getAdvertising();

NimBLEService *service = server->createService(UUID);

NimBLECharacteristic *characteristic = service->createCharacteristic(UUID, properties);
characteristic->setCallbacks(&ble_callbacks);

service->start();

advertising->setScanResponse(true);
advertising->start();
```

Staars, 2023-10-13:
There was a breaking change in the callback API some months ago. I had the same problem.

The ugly thing is, that you do not get a compile error. The old callbacks are „gracefully ignored“.

Just check the current header files.

h2zero, 2023-10-15:
Yes, this is unfortunately a compatibility breakage in the master branch. You'll need to check and change your function signatures.

jakubarm, 2023-10-16:
Yes, the function signature in the callback class was the problem. Thank you

closed 2023-10-16 by jakubarm