h2zero/esp-nimble-cpp #135: BLE Uart is not publishing service uuid when trying to connect with BLE client

Firmware-beginner, 2023-10-09:
Im running BLE Client code on one esp32s3 chip and BLE uart(server) code on another esp32s3 chip. The ble client shows the devices it discovers but its not connecting to ble uart. Upon further investigation it was found that the ble client is not able to see the service uuid of ble uart, not sure whether its the ble client issue or ble uart issue.

h2zero, 2023-10-10:
Looks like it's finding it, I'm not seeing anywhere that you're trying to connect though. Please post a log of the connection attempt for diagnosis.

Firmware-beginner, 2023-10-10:
Logs of Client (BLE client)
I (375) app_start: Starting scheduler on CPU0
I (379) app_start: Starting scheduler on CPU1
I (379) main_task: Started on CPU0
I (389) main_task: Calling app_main()
Starting BLE Client application...
I (409) BLE_INIT: BT controller compile version [85b425c]
I (409) phy_init: phy_version 601,fe52df4,May 10 2023,17:26:54
I (449) BLE_INIT: Bluetooth MAC: f4:12:fa:81:4c:5e

I (449) NimBLE: GAP procedure initiated: stop advertising.

I (449) NimBLE: GAP procedure initiated: discovery;
I (449) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (459) NimBLE: duration=5000ms
I (469) NimBLE:

I (469) main_task: Returned from app_main()
BLE Advertised Device found:  -
BLE Advertised Device found:  -
BLE Advertised Device found: DMP538F - bae55b96-7d19-458d-970c-50613d801bc9
BLE Advertised Device found:  - 0xfeaa
BLE Advertised Device found: FT_CURV_B184 - 0x5533
BLE Advertised Device found:  -
BLE Advertised Device found:  -
BLE Advertised Device found: FB BSW042 -
BLE Advertised Device found:  -
BLE Advertised Device found:  -
BLE Advertised Device found: UART Service -
BLE Advertised Device found:  -

Logs of Server (BLE uart)
I (375) app_start: Starting scheduler on CPU0
I (380) app_start: Starting scheduler on CPU1
I (380) main_task: Started on CPU0
I (390) main_task: Calling app_main()
I (410) BLE_INIT: BT controller compile version [85b425c]
I (410) phy_init: phy_version 601,fe52df4,May 10 2023,17:26:54
I (440) BLE_INIT: Bluetooth MAC: 34:85:18:8e:b6:a6

I (450) NimBLE: GAP procedure initiated: stop advertising.

I (450) NimBLE: GAP procedure initiated: advertise;
I (450) NimBLE: disc_mode=2
I (450) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (460) NimBLE:

Waiting a client connection to notify...
I (470) main_task: Returned from app_main()

h2zero, 2023-10-12:
I don't see anywhere in the logs where you've attempted to connect to the server device. You need to create a client instance and call the connect method with the server device you want to connect to.

Firmware-beginner, 2023-10-12:
Changes made in the BLE Client code : In MyAdvertisedDeviceCallbacks -> BLEAdvertisedDevicecallbacks its printing the device name and service UUID of newly discovered devices. It will connect to the server if it finds server's UUID. Its not printing the service UUID and further not executing the if statement.  I am not understanding why it is not showing the service UUID of the server to which the client wants to connect to.

I have added a printf statement to print the name and UUID of the devices the client will scan and there I see the name of the device i want to connect to but not service UUID. I believe the client will only initiate connection when it will find a matching UUID.

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */

/*** Only a reference to the advertised device is passed now
  void onResult(BLEAdvertisedDevice advertisedDevice) { **/
  void onResult(BLEAdvertisedDevice* advertisedDevice) {
//    printf("BLE Advertised Device found: %s\n", advertisedDevice->toString().c_str());
	  printf("BLE Advertised Device found: Name : %s - UUID : %s\n", advertisedDevice->getName().c_str(), advertisedDevice->getServiceUUID().toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
/********************************************************************************
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
********************************************************************************/
    if (advertisedDevice->haveServiceUUID() && advertisedDevice->isAdvertisingService(serviceUUID)) {

      BLEDevice::getScan()->stop();
/*******************************************************************
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
*******************************************************************/
      myDevice = advertisedDevice; /** Just save the reference now, no need to copy the object */
      doConnect = true;
      doScan = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

Below is the logs on the client side : 
I (469) main_task: Returned from app_main()
BLE Advertised Device found: Name : UART Service - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID : 0xfef3
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name : HONOR Band 5-346 - UUID :
BLE Advertised Device found: Name : FBBTS11 - UUID :
BLE Advertised Device found: Name : HONOR Band 6-A41 - UUID :
BLE Advertised Device found: Name : FT_CURV_B184 - UUID : 0x5533
BLE Advertised Device found: Name : Amazfit Bip Watch - UUID : 0xfee0
BLE Advertised Device found: Name :  - UUID : 0xfd69
BLE Advertised Device found: Name : FB BSW042 - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID :
BLE Advertised Device found: Name :  - UUID :

h2zero, 2023-10-15:
Yes, the code you are using is looking for specific advertisements and if found will set a flag to connect to that device. The log doesn't indicate it was found, so no connection attempt made. 

If I may suggest, using another esp32 make a server device with the example code and use the example client code as is and see how it connects.

h2zero, 2024-06-14:
Closing this as stale.

closed 2024-06-14 by h2zero