h2zero/esp-nimble-cpp #137: driver error on Windows

13Pixel, 2023-11-07:
Hello, I'm having some issues with HID and windows. On Linux everything work fine.

I modified Simple ble server example and added NimBLEHIDDevice. Problem is that on Windows I get driver error, on this bluetooth device. 

Does anyone had similar issues?

Thanks.

thomasMEBaker, 2024-01-24:
@13Pixel - I am experiencing the same issue. Did you find a fix for this?

13Pixel, 2024-02-06:
Unfortunately no, Couldn't find way to fix this, so put it on hold till I'll have time to figure it out.

h2zero, 2024-02-06:
Do you have bonding enabled? I believe it's a requirement in windows.

thomasMEBaker, 2024-02-06:
As in "Persist the BLE Bonding keys in NVS" within the Nimble Options?

h2zero, 2024-06-14:
You should enable that as well as set the device to bond by calling `NimBLEDevice::setSecurityAuth(true, false, true);` You'll also need to call `NimBLEDevice::startSecurity` in the on connect callback or set the characteristic properties to require encryption.

closed 2024-11-20 by h2zero