h2zero/esp-nimble-cpp #138: How to reconnect to a remote control after long idle time?

shooshx, 2023-11-11:
Thank you for this excellent library and for being so responsive to help!

I have a cheap remote control from Aliexpress that presents a HID service. I got to the point where I can connect to it when it's advertising and get notifications from it.

The only problem I have left is that after 2 minutes from press event, the device disconnects. If I understand correctly and then the only way to reconnect is to press the button combination to advertise again.  
From my understanding, once a connection is established, and then disconnects, on the next button press on the device it's supposed to send a "directed" advertising meant only for the paired host (the ESP32) so that the connection is restored. I was not able to see this happening.  
What am I missing?

my code: https://github.com/shooshx/led_matrix_clock/blob/main/ble/ble_remote/ble_remote.ino

shooshx, 2023-11-11:
Update: I just used the windows BT sniffer to see what packets are being sent from the remote control and some of what was going on there mentioned "LE Extended Advertising Report", which led me to read here:

https://github.com/h2zero/NimBLE-Arduino/blob/bc333ccb6e8d9ff2059af9cbd5006a290a4de3a5/src/nimconfig.h#L41

that Extended Advertising only works with ESP32C3, ESP32S3, ESP32H2, which I don't have
I guess it's back to Aliexpress to order one of these...

h2zero, 2024-06-14:
@shooshx did you resolve this?

shooshx, 2024-06-16:
No, my conclusion after having bought 3 such remotes was that BLE remotes come in 2 flavors, the ones that come with their own USB dongle and the ones that are on BLE5 which the base ESP32 doesn't support

closed 2024-11-20 by h2zero