h2zero/esp-nimble-cpp #140: Taking more time to connect with ESP32 BLE Server

demirschmid61, 2024-01-17:
Hello sir,

Thank you very much for this wonderful library which reduces the RAM. I am trying to migrate from ArduinoBLE to NimBLE, but facing some issues.

When I am trying to connect from NRF it's **taking more time to connect** to the ESP32 BLE server compared to ardroidBLE. I have 6 characteristics. 

The BLE initialization code snippet is mentioned below
`void initBLE()
{
  NimBLEDevice::init("ESP32Server");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);
  NimBLEDevice::setMTU(516);

  // Create Server
  pServer = NimBLEDevice::createServer();

  pServer->setCallbacks(new ServerCallBack()); // Set callback for onConnect onDisconnect

  // Create Service
  createService();
  NimBLEAdvertising *pAdvertising = pServer->getAdvertising();

  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->start();
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);
  pAdvertising->setMinPreferred(0x12);
  NimBLEDevice::startAdvertising();
}`

The service and characteristic creation code snippet is mentioned below

`void createService()
{
  NimBLEService *pService = pServer->createService(SERVICE_UUID);

  pCharacteristic1 = pService->createCharacteristic(CHARACTERISTIC_UUID1,
                                                    NIMBLE_PROPERTY::READ |
                                                        NIMBLE_PROPERTY::WRITE_NR |
                                                        NIMBLE_PROPERTY::WRITE |
                                                        NIMBLE_PROPERTY::NOTIFY);
  pCharacteristic2 = pService->createCharacteristic(CHARACTERISTIC_UUID2, NIMBLE_PROPERTY::READ |
                                                                              NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic5 = pService->createCharacteristic(CHARACTERISTIC_UUID5, NIMBLE_PROPERTY::READ |
                                                                        NIMBLE_PROPERTY::NOTIFY);
  pCharacteristic3 = pService->createCharacteristic(CHARACTERISTIC_UUID3, NIMBLE_PROPERTY::READ |
                                                                              NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic4 = pService->createCharacteristic(CHARACTERISTIC_UUID4, NIMBLE_PROPERTY::READ |
                                                                              NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic6 = pService->createCharacteristic(CHARACTERISTIC_UUID6, NIMBLE_PROPERTY::READ |
                                                                            NIMBLE_PROPERTY::WRITE_NR |
                                                                            NIMBLE_PROPERTY::WRITE |
                                                                            NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic7 = pService->createCharacteristic(CHARACTERISTIC_UUID7, NIMBLE_PROPERTY::READ |
                                                                                    NIMBLE_PROPERTY::WRITE_NR |
                                                                                    NIMBLE_PROPERTY::WRITE |
                                                                                    NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic8 = pService->createCharacteristic(CHARACTERISTIC_UUID8, NIMBLE_PROPERTY::READ |
                                                                    NIMBLE_PROPERTY::WRITE_NR |
                                                                    NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic9 = pService->createCharacteristic(CHARACTERISTIC_UUID9, NIMBLE_PROPERTY::READ |
                                                                    NIMBLE_PROPERTY::WRITE_NR |
                                                                    NIMBLE_PROPERTY::NOTIFY);

  pCharacteristic1->setCallbacks(&chrCallBack);       
  pCharacteristic6->setCallbacks(&confCallBack); 
  pCharacteristic7->setCallbacks(&synchCallback);
  pService->start();
}`

Is there anything i am missing?

[image]

h2zero, 2024-01-17:
You're quite welcome.

The issue could just be the nRF app, there should be no issue on the NimBLE side that can cause this.

h2zero, 2024-06-14:
@demirschmid61 have you tried adjusting the advertising interval?