h2zero/esp-nimble-cpp #142: esp-idf v5.2 build issues due to missing #include <ctype>

cmorganBE, 2024-02-23:
Reporting here, will open a PR tomorrow for the issue if someone else doesn't fix it first.

Patch to fix:

```
diff --git a/src/NimBLEAttValue.h b/src/NimBLEAttValue.h
index be346d5..786bc8b 100644
--- a/src/NimBLEAttValue.h
+++ b/src/NimBLEAttValue.h
@@ -24,6 +24,7 @@
 
 #include <string>
 #include <vector>
+#include <ctime>
 
 #ifndef CONFIG_NIMBLE_CPP_ATT_VALUE_TIMESTAMP_ENABLED
 #    define CONFIG_NIMBLE_CPP_ATT_VALUE_TIMESTAMP_ENABLED 0
```

```
/home/cmorgan/projects/x/firmware/components/esp-nimble-cpp/src/NimBLEAttValue.h:178:5: error: 'time_t' does not name a type
  178 |     time_t          getTimeStamp() const   { return 0; }
      |     ^~~~~~
/home/cmorgan/projects/x/firmware/components/esp-nimble-cpp/src/NimBLEAttValue.h:19:1: note: 'time_t' is defined in header '<ctime>'; did you forget to '#include <ctime>'?
   18 | #include "NimBLELog.h"
  +++ |+#include <ctime>
   19 | 
/home/cmorgan/projects/x/firmware/components/esp-nimble-cpp/src/NimBLEAttValue.h:180:34: error: 'time_t' has not been declared
  180 |     void            setTimeStamp(time_t t) { }
      |                                  ^~~~~~
/home/cmorgan/projects/x/firmware/components/esp-nimble-cpp/src/NimBLEAttValue.h:203:30: error: 'time_t' has not been declared
  203 |     const uint8_t*  getValue(time_t *timestamp);
      |                              ^~~~~~
/home/cmorgan/projects/x/firmware/components/esp-nimble-cpp/src/NimBLEAttValue.h:257:18: error: 'time_t' has not been declared
  257 |     T   getValue(time_t *timestamp = nullptr, bool skipSizeCheck = false) {

```

h2zero, 2024-02-25:
Thank you for brining this up, I'd be happy to accept a PR.

kiss81, 2024-03-01:
> Thank you for brining this up, I'd be happy to accept a PR.

I made a PR for it.

cmorganBE, 2024-05-09:
Fixed in #144. Thanks @kiss81

closed 2024-05-09 by cmorganBE