h2zero/esp-nimble-cpp #148: Random crash on launch ESP-IDF 5.2.1

mmackh, 2024-03-18:
Hello,

I'm trying to track down a crash that happens randomly on launch somewhere in the BLE stack with the latest release of ESP-IDF (5.2.1) and and ESP32-WROOM-32E. Attached are the crash reports - it's both random which one causes the system to reboot and when. So it sometimes get stuck on boot for a few times and then it goes through ok.

```
EXCVADDR: 0x84c2837a  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Backtrace: 0x400e95ec:0x3ffbb570 0x400de3e2:0x3ffbb590 0x400dc596:0x3ffbb5d0 0x400da396:0x3ffbb610 0x400da4f3:0x3ffbbe50 0x401a43ce:0x3ffbbe70

0x400e95ec: ble_gatts_cpfd_is_sane at ble_gatts.c:751
   (inlined by) ble_gatts_count_resources at ble_gatts.c:3103
     (inlined by) ble_gatts_count_cfg at ble_gatts.c:3137
      0x400de3e2: NimBLEService::start() at NimBLEService.cpp:235
        0x400dc596: Sentionic::BluetoothLE::startHost(char const*, char const*, char const*, std::function) at BluetoothLE.cpp:103
          0x400da396: boot() at main.cpp:124 (discriminator 1)
            0x400da4f3: app_main at main.cpp:247
              0x401a43ce: main_task at app_startup.c:208
              
```

```
abort() was called at PC 0x400dddad on core 0
Backtrace: 0x40081732:0x3ffbb470 0x4009290d:0x3ffbb490 0x400982e2:0x3ffbb4b0 0x400dddad:0x3ffbb520 0x400dc746:0x3ffbb570 0x400dd56a:0x3ffbb5b0 0x400dc5b1:0x3ffbb5d0 0x400da396:0x3ffbb610 0x400da4f3:0x3ffbbe50 0x401a43ce:0x3ffbbe70

0x40081732: panic_abort at panic.c:472
  0x4009290d: esp_system_abort at esp_system_chip.c:93
    0x400982e2: abort at abort.c:38
      0x400dddad: NimBLEServer::start() at NimBLEServer.cpp:191
        0x400dc746: NimBLEAdvertising::start(unsigned long, void (*)(NimBLEAdvertising*), NimBLEAddress*) at NimBLEAdvertising.cpp:419
          0x400dd56a: NimBLEDevice::startAdvertising(unsigned long) at NimBLEDevice.cpp:181 (discriminator 1)
            0x400dc5b1: Sentionic::BluetoothLE::startHost(char const*, char const*, char const*, std::function) at BluetoothLE.cpp:111
              0x400da396: boot() at main.cpp:124 (discriminator 1)
                0x400da4f3: app_main at main.cpp:247
                  0x401a43ce: main_task at app_startup.c:208
```

mmackh, 2024-03-18:
The crash seems to occur only when a 128bit UUID is assigned to the characteristic. 

For example, this crashes:
```5ad7909f-c841-44c6-8460-9a1143d2bf98```
but this is ok:
```0x2B19```

h2zero, 2024-03-24:
Thanks, seems espressif has added this strange new feature that is not in the official NimBLE code. Not sure if simply checking the IDF version will be enough to detect the existence of this.

mmackh, 2024-03-25:
I thought a little about the reason why something like this could happen and about C++ (still quite new for me): 

https://stackoverflow.com/questions/5914422/proper-way-to-initialize-c-structs

Early test suggest this could be a fix:

` p C h r _ a   =   n e w   b l e _ g a t t _ c h r _ d e f [ n u m C h r s   +   1 ] ( ) ;`

holdersn, 2024-03-26:
yes, I think initializing the instances, as you suggested, should fix this issue in a backwards-compatible way.

It worked fine for me and I've prepared a pull request, #150

commit 226c67f7 (2024-04-20): Initialize ble_gatt_chr_def[]. Fixes #148 (#150)

IDF 5.2 introduced a new member, cpfd, to the
ble_gatt_chr_def struct. It needs to be initialized
to nullptr in order to avoid accessing uninitialized
memory. By initializing the whole struct, we get
everything initialized in a backward-compatible way.