h2zero/esp-nimble-cpp #151: NimBLEService::start() causes LoadProhibited error when using external SPI Ram

chrispbarlow, 2024-03-27:
Since external RAM is not zeroed at startup, an overflow occurs in ble_gatts_count_resources.

To resolve this, `ble_gatt_svc_def* svc`, `ble_gatt_chr_def* pChr_a` and `ble_gatt_dsc_def* pDsc_a` need to be initialised to zero in NimBLEService::start() 

The following patch fixes this (based on top of release/1.4 branch):

```
diff --git a/src/NimBLEService.cpp b/src/NimBLEService.cpp
index 0ae2e9f..4bb6fa5 100644
--- a/src/NimBLEService.cpp
+++ b/src/NimBLEService.cpp
@@ -127,7 +127,7 @@ bool NimBLEService::start() {
         // Nimble requires an array of services to be sent to the api
         // Since we are adding 1 at a time we create an array of 2 and set the type
         // of the second service to 0 to indicate the end of the array.
-        ble_gatt_svc_def* svc = new ble_gatt_svc_def[2];
+        ble_gatt_svc_def* svc = new ble_gatt_svc_def[2]{};
         ble_gatt_chr_def* pChr_a = nullptr;
         ble_gatt_dsc_def* pDsc_a = nullptr;
 
@@ -160,7 +160,7 @@ bool NimBLEService::start() {
             // Nimble requires the last characteristic to have it's uuid = 0 to indicate the end
             // of the characteristics for the service. We create 1 extra and set it to null
             // for this purpose.
-            pChr_a = new ble_gatt_chr_def[numChrs + 1];
+            pChr_a = new ble_gatt_chr_def[numChrs + 1]{};
             uint8_t i = 0;
             for(auto chr_it = m_chrVec.begin(); chr_it != m_chrVec.end(); ++chr_it) {
                 if((*chr_it)->m_removed > 0) {
@@ -189,7 +189,7 @@ bool NimBLEService::start() {
                     pChr_a[i].descriptors = NULL;
                 } else {
                     // Must have last descriptor uuid = 0 so we have to create 1 extra
-                    pDsc_a = new ble_gatt_dsc_def[numDscs+1];
+                    pDsc_a = new ble_gatt_dsc_def[numDscs+1]{};
                     uint8_t d = 0;
                     for(auto dsc_it = (*chr_it)->m_dscVec.begin(); dsc_it != (*chr_it)->m_dscVec.end(); ++dsc_it ) {
                         if((*dsc_it)->m_removed > 0) {
```

h2zero, 2024-04-21:
Thanks, I will look to change this soon.

michaelboeding, 2024-04-21:
I'm also running into this issue.

joshbillions, 2024-05-09:
Same issue here. @chrispbarlow 's fix resolves it.

h2zero, 2024-06-14:
Resolved with #148

closed 2024-06-14 by h2zero