h2zero/esp-nimble-cpp #154: Modernizing the codebase

mickeyl, 2024-04-08:
I wonder whether it would be feasible to update the codebase a bit with regards to modern C++ constructs.

In particular, esp-nimble-cpp relies a lot on pointers (e.g., for callbacks). Devices such as `std::unique_ptr` and `std::shared_ptr` could improve this a lot.

But also trivial things like using `auto` and `for (auto& item: collection)` could help to make the codebase more simple to read, more friendly towards contributions, and better maintainable.

What's your take on that?

h2zero, 2024-04-20:
Yes I've wanted to do some cleanup as the code is based on a pretty old library now. I just haven't had time or any real need for it.

mickeyl, 2025-05-23:
I guess this issue is a bit pointless after all. First, you want to support older compilers as well, and 2nd, you will accept pull requests improving the situation. I guess it's better to just close this here, sorry for the noise.

h2zero, 2025-05-25:
I'm not opposed to modernizing this in the future, it's just that the codebase is shared with Arduino so it would break a lot of projects with using any of the newer features for a bit. I'm also of the mind that some of the new stuff's implementation will increase resource use unnecessarily so it would need to be checked. Thanks for bringing this up!