h2zero/esp-nimble-cpp #155: deleteAllBonds() not properly cleaning up space in NVS

finger563, 2024-05-03:
@h2zero i've noticed (on esp-idf `v5.2.1`) that the `ble_store_clear()` doesn't work as expected - I still get too many bonded device errors even though every time i enter pairing mode I call `NimBLEDevice::deleteAllBonds()`.

With logs enabled:
[image]

_Originally posted by @finger563 in https://github.com/h2zero/esp-nimble-cpp/issues/5#issuecomment-2093397724_

h2zero, 2024-05-03:
That would be an upstream issue. Espressif probably changed something in the NimBLE code and broke it.

finger563, 2024-05-03:
FWIW i've found that manually deleting the bonds 1-1 is a better approach for now, though it does sometimes actually return errors. Currently the `NimBLEDevice::deleteAllBonds()` does not return any error code (though it could since the underlying call does). Perhaps I'll make a PR for that to help track errors.

Some of my testing can be seen here:
https://github.com/esp-cpp/espp/pull/225

finger563, 2024-05-04:
Somewhat related, it can be useful to see the return code from the underlying `ble_store_clear()`, so I've made a PR: #158

h2zero, 2024-06-14:
Closing this as #158 was merged and the issue is upstream.

closed 2024-06-14 by h2zero