h2zero/esp-nimble-cpp #159: Bonded centrals not able to reconnect

cmorganBE, 2024-05-15:
I have a device that I'd like to run in two modes:

- Previously bonded centrals can connect only
- New devices can pair/connect/bond

This is controlled by a push button with the intent to protect the device from remote tampering by requiring physical interaction with the device to pair/connect/bond new devices.

What I've done so far:
- Switched characteristics over to 'READ | READ_ENC' or 'WRITE | WRITE_ENC'
- NimBLEDevice::setSecurityAuth(true, false true); // bonding = true, mitm = false, secure connection = true
- Enabled Nimble persisting bonding to NVS
- NimBLEAdvertising::setScanFilter(false, false); // at the moment I'm not using whitelists

What I see:
- iphone asks to pair when I open a characteristic on lightblue
- Pairing works, characteristics respond
- Disconnect iphone
- Reconnect iphone
- Connections fail (nothing useful on the esp32 side), lightblue says "failed to connect"

If I want to reconnect I have to go to Settings->Bluetooth and 'Forget' the device.

Using nvs_get I'm seeing the bond in nvs as 'local_irk_1', so that's being persisted.

Is the approach or using setScanFilter() the right one for controlling the connection modes?
Is there something I'm missing on the device side that is preventing reconnections using the bond?

dorianim, 2024-05-16:
Seems to be the same issue as I have, see:
https://github.com/h2zero/esp-nimble-cpp/issues/11#issuecomment-2116145784

h2zero, 2024-06-14:
This should be resolved now via #164

closed 2024-06-14 by h2zero